cyberbotics/webots #1529: Bug in projects/samples/devices/worlds/pen.wbt

chinakwy, 2020-04-08:
**Describe the Bug**
Hello, I'm using "pen" to draw trail. but I'm confused ,when I change the size of the arena, why does the trail  disappear and "pen" no longer draws trail ?
**Steps to Reproduce**
1. Go to 'projects/samples/devices/worlds/pen.wbt '
2. Click on 'RectangleArena (PROTO)'
3. Scroll down to 'floorSize'，and change it
4. See error：the trail  disappear and "pen" no longer draws trail

**Expected behavior**
A clear and concise description of what you expected to happen.
the trail should be still there，and "pen" can continue draw trail

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]
**System**
 - Operating System: macOS Catalina
 - Graphics Card: Intel Iris Plus Graphics 640 1536 MB

**Additional context**
Add any other context about the problem here.

omichel, 2020-04-08:
You should save the world after your change the size of the floor and reload it, so that the pen can work again with the resized floor. Let me know if that resolves the problem.

chinakwy, 2020-04-17:
Hallo thanks for you reply

but it still works not well

 original 
[image]
when I change the size, it looks like below
[image]

omichel, 2020-04-17:
This is because should should also adjust the texture mapping of the floor when you change its size. The [Floor](https://github.com/cyberbotics/webots/blob/master/projects/objects/floors/protos/Floor.proto) PROTO doesn't handle this automatically because it uses repeating textures when scaled up. So when changing the `size` of this PROTO, you should also change the `tileSize` in the very same way, e.g., if you have `size 1 2`, you should also have `tileSize 1 2`. Similarly if you are using the [RectangleArena](https://github.com/cyberbotics/webots/blob/master/projects/objects/floors/protos/Floor.proto) PROTO, you should have matching `floorSize` and `floorTileSize`. You may need to create your own custom tile so that the floor resolution and content match your needs.

chinakwy, 2020-04-17:
Thanks! It has done

closed 2020-04-17 by chinakwy