cyberbotics/webots #1534: "Track" node documentation missing

telramund, 2020-04-09:
**Documentation page(s) concerned**
https://www.cyberbotics.com/doc/reference/node-chart
https://www.cyberbotics.com/doc/reference/track

**What is the problem with the current documentation?**
Go to the node chart and click "Track". Instead of description of the "Track" node, the documentation front page is shown. I believe other links to "Track" are broken in this way too. The page is there as an .md file though.

**Describe how you would fix it**
Not sure. I am not familiar with documentation generation process yet.

DavidMansolino, 2020-04-09:
Which browser/OS are you using?
For me it seems to work fine on Firefox / Ubuntu 18.04
Can you also try to clean your cache and retry?

About the documentation, there is no generation process, the md files are directly interpreted on the fly be the documentation system, for example, for the no chart page, this is in this md file:
https://github.com/cyberbotics/webots/blob/master/docs/reference/node-chart.md

Note that this page is a quite complex one, the one of the track for example is way simpler: 
https://github.com/cyberbotics/webots/blob/master/docs/reference/track.md

omichel, 2020-04-09:
FYI, I tested on Windows 10 with Edge, Chrome and Firefox and it worked in all the cases.

telramund, 2020-04-09:
Looks like it's because of ad blocker in Pale Moon browser. Everything is fine in Firefox with ad blocker on, or in Pale Moon without one. Never saw an adblocker mess things up in this way, sorry for false alarm.

closed 2020-04-09 by telramund

DavidMansolino, 2020-04-10:
Thank you for the feedback.