cyberbotics/webots #1536: "light strip" type for Parquetry appearance not working

telramund, 2020-04-13:
**Describe the Bug**

Specifying "light strip" type for Parquetry appearance gives the same result as "dark strip", contrary to the documentation:
https://www.cyberbotics.com/doc/guide/appearances#parquetry

**Steps to Reproduce**

Create a world file with the following content:

```
#VRML_SIM R2020a utf8
WorldInfo { }
TexturedBackground { }
TexturedBackgroundLight { }
Floor {
  size 50 50
  appearance Parquetry { type "light strip" }
}
```
Observe that floor appearance matches "dark strip", not "light strip". Change type to "dark strip", note that the appearance has not changed.

**System**
 - Operating System: Gentoo Linix
 - Graphics Card: NVIDIA 940MX

DavidMansolino, 2020-04-14:
You are completely right, it seems their is a bug with our online documentation, the documentation of the next release of Webots is online and not of the current one, please find the correct version of the documentation here:

https://www.cyberbotics.com/doc/guide/appearances?version=R2020a-rev1

omichel, 2020-07-17:
This is because changes for Webots R2020a revision 2 were committed on the master branch and got automatically published as the default online documentation. We should have a created a branch for revision 2 instead. We will take are of this from Webots R2020b revision 1.

closed 2020-07-17 by omichel