cyberbotics/webots #1538: Problem using group as boundingobject

DengYuelin, 2020-04-14:
I have a solid with a group as its children, and the group as the boundingobject. The group has contained two transforms, the first has a children shape DEF A, the second children use DEF A, (two identical shape, cylinder, at different place). When I change the radius of A, both cylinders change shape, which is correct. However, only one side of the boundingobject(white border) changed, resulting in different boundingobject on two sides. See the picture for detail.
[image]

If I save and reload, the two sides will match. However, I intended to change the radius using a program during simulation, so S&L isn't an option.

**System**
I tried this on both systems:
1, Ubuntu 16.04 with webots 2020a-rev1;
2, Windows 10 with webots 2019b-rev1;

**Additional context**
The world I'm using.
[Double_peg_hole.zip](https://github.com/cyberbotics/webots/files/4473560/Double_peg_hole.zip)

stefaniapedrazzi, 2020-04-15:
Thank you for reporting this bug.
I'm inspecting it and I will fix it as soon as possible.

Meanwhile there are some workarounds.
For example if you use this structure, then resizing the peg works correctly:

```
Solid { 
  childen [
    DEF PEG1 Transform {
      children [
        DEF PEGSHAPE Shape {
          ...
        }
      ]
    }
    DEF PEG2 Transform {
      translation 0.04 0 0
      children [
        USE PEGSHAPE
      ]
    }
  ]
  boundingBox Group {
    children [
      USE PEG1
      Transform {
        translation 0.04 0 0
        children [
          USE PEGSHAPE
        ]
      }
    ]
  }
}
```

stefaniapedrazzi, 2020-05-18:
This issue is fixed in #1544.

closed 2020-05-18 by stefaniapedrazzi