cyberbotics/webots #1542: Optimize Sound Engine

DavidMansolino, 2020-04-15:
**Describe the Bug**
Currently the sound engine is initialised when Webots is started, but in 99% of the cases this is not needed (sound is disabled by default).

This is therefore not efficient because it takes time and consume resources uselessly, furthermore sound cause many disturbing 'ALSA' warnings in Webots on computer without sound devices (on travis for example, but also reported by many users).

Instead, we should initialise the sound engine only the first time it is actually used.

sigmunau, 2021-09-23:
Is there some way to run webots without the sound engine if you know you don't need it, and the alsa warnings are annoying you? Maybe a command line switch could work?

omichel, 2021-09-30:
Yes, a command line switch seems to be a good idea and could be implemented probably fairly easily.

closed 2021-11-15 by BenjaminDeleze