cyberbotics/webots #1549: Robot with <extern> controller does not appear in streaming viewer

ecervera, 2020-04-16:
**Describe the Bug**
A robot with an `<extern>`  controller is not shown in the streaming viewer. Chrome shows only the background environment, and Firefox shows just a black window.

**Steps to Reproduce**
1. Open Webots with `--stream` and a sample world, e.g.
```
$WEBOTS_HOME/webots --stream $WEBOTS_HOME/projects/robots/adept/pioneer3/worlds/pioneer3dx.wbt 
```
or
```
$WEBOTS_HOME/webots --stream $WEBOTS_HOME/projects/robots/k-team/khepera-4/worlds/khepera4.wbt 
```
2. Open the streaming viewer in Chrome/Firefox, by opening the file `$WEBOTS_HOME/resources/web/streaming_viewer/index.html`
3. Click on the connect button: you will see the simulation running both in Webots and the streaming viewer.
4. Pause the simulation in Webots.
5. Reload world.
6. Open the node `Pioneer3dx` / `Khepera4` in the Scene Tree and click on the controller field.
7. Click on the Select... button and pick the `<extern>` controller from the list.
8. Save a copy of the world (File / Save World as...) e.g. in the desktop.
9. Reload World (the scene is reloaded in the streaming viewer but the robot does not appear).
10. Click on the Select... button and pick the `void` controller from the list.
11. Save and reload world (the scene is reloaded in the streaming viewer and the robot appears again).

The behavior is the same no matter the synchronization field is FALSE or TRUE.

**Expected behavior**
The robot should appear in the streaming viewer.

**System**
 - Operating System: Linux Ubuntu 18.04
 - Graphics Card: NVIDIA Quadro P3000
 - Webots R2020a rev1
 - Browser: Chrome  81.0.4044.92,  Firefox 75.0

**Additional context**
I am developing external controllers in Python and they work fine, except that the robot does not show in the streaming viewer.

DavidMansolino, 2020-04-16:
We fixed a similar issue recently: #1305
May I ask you to try with a nightly build of Webots R2020a-revision2: https://github.com/cyberbotics/webots/releases

ecervera, 2020-04-16:
Thanks, the problem is solved with the nightly build 15_04_2020 :-)

closed 2020-04-16 by ecervera

DavidMansolino, 2020-04-16:
Perfect, thank you for the feedback!