cyberbotics/webots #1550: ROS service supervisor_get_from_def won't work with webots/develop

Acwok, 2020-04-16:
I'm using the develop branch of Webots and the ros-kinetic-webots-ros package.
I get the following error when using the supervisor_get_from_def service from ROS kinetic:
```
[ros] [ERROR] [1587023118.216799425, 4.456000000]: client wants service /tile_35_4366_pc047/supervisor/get_from_def to have md5sum ac26007a2c83bd1b38318cda0f4ce627, but it has 3f818aa8f7c2c60588c99f7ee189c5bd. Dropping connection.
```

Acwok, 2020-04-16:
@DavidMansolino it was working with [Webots R2020a revision1 .deb](https://github.com/cyberbotics/webots/releases/tag/R2020a-rev1)     

I've made a diff between the two versions:  
- webots/projects/default/controllers/ros/include/webots_ros/supervisor_get_from_def.h
- webots/projects/default/controllers/ros/include/webots_ros/supervisor_get_from_defRequest.h
- webots/projects/default/controllers/ros/include/webots_ros/supervisor_get_from_defResponse.h

Here are the diff: [webots_diff.zip](https://github.com/cyberbotics/webots/files/4487341/webots_diff.zip)

Acwok, 2020-04-16:
@DavidMansolino, the ROS service is working fine with webots/master

DavidMansolino, 2020-04-17:
I checked what is happening, and indeed the definition of the `supervisor_get_from` service has changed on the develop branch, you can't therefore use the develop branch with the released version of `ros-kinetic-webots-ros` (which is made to work with the latest release of Webots). If you want to use Webots from the develop branch, you have to compile the `webots-ros` package by yourself using the files from here: https://github.com/cyberbotics/webots/tree/develop/projects/languages/ros/webots_ros
And the services and messages from here:
https://github.com/cyberbotics/webots/tree/develop/projects/default/controllers/ros/include

closed 2020-04-21 by DavidMansolino