cyberbotics/webots #1565: Open source worlds

stevemartinov, 2020-04-17:
I tried to find the worlds (for example big warehouse or a factory). Any idea where can I find that?

remi-cyberbotics, 2020-04-17:
@stevemartinov Please have a look on our documentation: https://www.cyberbotics.com/doc/guide/sample-webots-applications

(Access the samples through `File` then `Open sample world..`)

closed 2020-04-17 by remi-cyberbotics

stevemartinov, 2020-04-17:
Yes I know that, I did read the documentation. If you would read my question, I meant big warehouse and factory. For example, in Unity that is achievable as they have a quite nice editor but what is for the Webots? Do we really need to manually add walls and objects from the editor? Does not sound efficient and time consuming.

remi-cyberbotics, 2020-04-20:
I did read your question, and as the documentation points out, all our sample worlds are located through the dedicated menu, there are no extra worlds elsewhere. However if you want to do this, I would suggest you to use the already existing Webots PROTO nodes, the graphical user interface and especially the [Def-Use mechanism](https://cyberbotics.com/doc/guide/tutorial-2-modification-of-the-environment#def-use-mechanism) (which will allow you to create one new object and then "copy-paste" it to populate your world), then you should achieve your goal pretty quick.