cyberbotics/webots #1566: Security issue: Unsafe temporary lib directory in linux launcher

MaxGaukler, 2020-04-17:
If I read the script correctly, the linux launcher is vulnerable to a library path injection:

https://github.com/cyberbotics/webots/blob/69202baee3c0974f69f0e72d3412d19ed69feaad/src/webots/launcher/webots-linux.sh#L64-L72

The temp dir is silently accepted if it already exists, even if it was created by another user. Therefore one user can escalate permissions to any other user launching webots on the same system.

Sketch of exploit:
- create /tmp/webots-lib-dir{1-9999999}
- place some .so file there that is used by webots and would otherwise be loaded from /usr/lib.
- modify that .so file to run exploit code
- make directory world-writable so that webots launcher doesn't fail when writing to path

Sketch of solution:
```
# safely create a temporary directory.
# Note that the following two lines cannot be merged into one because `export` would "hide" the return status of `mktemp`.
WEBOTS_TMP_PATH="$(mktemp -d $TMPDIR/webots-XXXXXX)" || exit 1
export WEBOTS_TMP_PATH
TMP_LIB_DIR="$WEBOTS_TMP_PATH/lib"
mkdir -p $TMP_LIB_DIR
```
At first glance, this sketch should work, however, `snap` and other edge cases should be double-checked.

omichel, 2020-04-20:
Thank you for this contribution. Please go ahead and open a pull request for that. We will be happy to review it.

MaxGaukler, 2020-04-20:
There are a few open issues that my PR doesn't yet fix:
- projects/samples/contests/ratslife/controllers/contest_manager/texture_generator.c uses unsafe TMPDIR; should switch to mktdemp(), QTemporaryDir or similar.
- MacOS should be checked, I'd guess the launcher and WbStandardPaths::webotsTmpPath()  are vulnerable
- WbStandardPaths::webotsTmpPath()  has an unsafe fallback for Linux
- WbControledWorld has the possibility of collisions, it uses $TMPDIR/webots_$PID_$random as filename without checking existence (or if it does, there's a TOCTOU race condition). I'm not sure about the semantics of socket files, if there's anything evil I can do if someone else creates a socket with predictable path. Maybe use it for an overwriting attack by placing a symlink?
- Maybe drop all the details in src/lib/Controller/api/system.c and just use QTemporaryDir (with autoRemove=false) or similar?

omichel, 2020-04-20:
Thank for this in depth analysis which looks very relevant.

The socket server is used to communicate with controllers, so it's not critical from a security point of view. However, I agree there is a risk of collision with the random number. A problem is that these sockets correspond to files with various locations (depending on the system), so it's difficult to check the file beforehand.

It's not easy to use QTemporaryDir in `src/lib/Controller/api/system.c` because this library (libController) has no dependency on Qt.

But anyhow, your PR is an improvement on the current version, so it's good to take.

We will review it as soon as you complete the 3 remaining tasks and mark it ready for review.

closed 2020-04-24 by DavidMansolino