cyberbotics/webots #1574: E-Puck Distance Sensors are not Correctly Modeled

lukicdarkoo, 2020-04-21:
I observed an inconsistent behaviour of the ToF sensor on the e-puck2 on the physical and simulated robot while making ROS2 node. Field Of View (aperture) is not configured for the ToF. After checking the proto I noticed that FoV is not configured for infra-red sensors either.

**ToF**: According to documentation FoV of ToF sensor used on the e-puck2 is 25 degrees:
http://projects.gctronic.com/epuck2/doc/VL53L0X-Datasheet.pdf (page 25)
while in the simulation the ToF sensor uses the default aperture (90 degrees). Also, `numberOfRays` should be configurable as it is done for infra-red sensors. 

**Infra-red**: Again, the aperture is not configured. According to GCtronic, it is 15 degrees:
https://github.com/gctronic/epuck_driver_cpp/blob/b728d28ca2986dc17030f3d01c3b869d49592d17/src/epuck_driver_cpp.cpp#L1046
but this should be verified. However, I am not sure how to estimate it.

Furthermore, default value of `distance_sensor_numberOfRays` is 1:
https://github.com/cyberbotics/webots/blob/master/projects/robots/gctronic/e-puck/protos/E-puck.proto#L27
I assume the value is chosen to increase performance, but we maybe should consider increasing it if the performance impact is not too high.

DavidMansolino, 2020-04-21:
> Furthermore, default value of distance_sensor_numberOfRays is 1

Yes this is indeed for performances reason, when the e-puck was calibrated, the infra-red distance sensors were the most computationally expensive part (even more than the low resolution camera of e-puck 1). But maybe with the recent improvements of Webots this is not the case anymore.

lukicdarkoo, 2020-04-21:
I made a mistake regarding the infra-red sensors. They have the aperture configured:
https://github.com/cyberbotics/webots/blob/914b38348a031c2856a0a2a85cf4c88ee5f8785b/projects/robots/gctronic/e-puck/protos/E-puckDistanceSensor.proto#L68

@DavidMansolino, I see that you chose `0.1` as value for the aperture. Do you think if it is more accurate than one configured by GCtronic:
https://github.com/gctronic/epuck_driver_cpp/blob/b728d28ca2986dc17030f3d01c3b869d49592d17/src/epuck_driver_cpp.cpp#L1046

DavidMansolino, 2020-04-21:
> @DavidMansolino, I see that you chose `0.1` as value for the aperture. Do you think if it is more accurate than one configured by GCtronic:
> https://github.com/gctronic/epuck_driver_cpp/blob/b728d28ca2986dc17030f3d01c3b869d49592d17/src/epuck_driver_cpp.cpp#L1046

I remember choosing this value because in my test scenario this was the value giving the most faithful result (but it probably depends a lot on the scenario).

lukicdarkoo, 2020-04-27:
Fixed in #1577

PR #1577 (2020-04-24): Add aperture to ToF sensor
**Description**
Models the distance sensors better (most noticeably ToF) ~~and allows configurable aperture~~.

**Related Issues**
Fixes #1574

**Tasks**
  - [x] Add aperture field to ToF sensors
  - [x] ~~Allow user to configure ToF and infra-red sensors (not sure if it is a good idea)~~
  - [x] Update documentation
closed 2020-04-27 by lukicdarkoo