cyberbotics/webots #1579: CameraRecognitionObject cannot return a correct value

90511, 2020-04-21:
**Describe the Bug**
Hello, I am confused with the all the functions in class CameraRecognitionObject in python which always return 0, but I have set  Recognition node correctly, and I can see rectangles surround the recognized objects.
**Steps to Reproduce**
1. I modified the code give in Tutorial 6: 4-Wheels Robot
2. add objects to the world 
3. the camera can recognize the objects but the all the founctions of CameraRecognitionObject returns zero
**code**
`from controller import Camera,CameraRecognitionObject`
`TIME_STEP = 64`
`robot = Robot()`
`camera = robot.getCamera('camera')`
`reco=CameraRecognitionObject()`
`camera.enable(TIME_STEP)`
`camera.recognitionEnable(TIME_STEP)`
`while robot.step(TIME_STEP) != -1:  `
 `       print('print the id of objects',reco.get_id())`
**Expected behavior**
The picture shows the id of apple is 398, but it reco.get_id() returns zero.

**Screenshots**
[image]
[image]

**System**
 - Operating System: Windows 10
 - Graphics Card: [e.g., NVIDIA GeForce RTX 1060

**Additional context**
I am not quite sure whether it is a bug in webots or my compling error.  Looking forward to your reply.

DavidMansolino, 2020-04-21:
Your use of the recognition API is wrong, you should not create a ``CameraRecognitionObject``object but rather get them from the camera with the ``camera.getRecognitionObjects`` function: https://cyberbotics.com/doc/reference/camera?tab-language=python#wb_camera_recognition_get_number_of_objects

Please have a look at this sample (it is in C but the use in Python is very similar):
https://github.com/cyberbotics/webots/blob/master/projects/samples/devices/controllers/camera_recognition/camera_recognition.c

Here are in particular the interesting part:
https://github.com/cyberbotics/webots/blob/master/projects/samples/devices/controllers/camera_recognition/camera_recognition.c#L56-L61

closed 2020-04-21 by DavidMansolino

90511, 2020-04-21:
Thanks for your immediate reply! Now I have output the information of the recognized data, but except id, others are returned as a Swig Object. 
[image]
Could you give me some instructions about reading the data?

DavidMansolino, 2020-04-22:
You have to use these functions to get the recognised data: https://cyberbotics.com/doc/reference/camera?tab-language=python#wb_camera_has_recognition

90511, 2020-04-22:
Thanks a lot for your immediate reply and help!  Now, I have perfectly solved my promble. :)

DavidMansolino, 2020-04-22:
You're welcome :-)