cyberbotics/webots #1598: CI source tests are failing on Appveyor if no PR was created

omichel, 2020-04-27:
Example: https://ci.appveyor.com/project/omichel/webots/builds/32472803
Probably related to #1590.

DavidMansolino, 2020-04-27:
Yes, this is clearly related to #1590: https://github.com/cyberbotics/webots/pull/1590/files#diff-5cc646629e8f88c2fe36e4f5434a24fbR75
Previously, if no PR was open the tests were silently ignore and raising success without testing anything (which looks extremely dangerous).
Unfortunately, we don't have a lot of solutions for this:
  - Failing the test: it is really easy to restart the test once the PR it open.
  - Marking the test as success: very dangerous as we don't know that the test didn't actually run.
  - Running the complete test on all the source files: This will congest our CI.

lukicdarkoo, 2021-03-03:
We don't use Appveyor anymore, we can close this.

closed 2021-03-03 by lukicdarkoo