cyberbotics/webots #1601: Unable to start Webots

Jajafarov, 2020-04-29:
**Describe the Bug**
I get a segmentation fault when trying to start Webots. 

**Steps to Reproduce**
1. Install Webots R2020a rev1 using snap.
2. Launch Webots.
4. See error

**Expected behavior**
Webots should launch.

**System**
 - Operating System: Arch Linux
 - Graphics Card: NVIDIA GeForce RTX 2070 Super 8 GB

**Additional context**
This is the specific error I get.
```
Fontconfig`` warning: FcPattern object weight does not accept value [40 210)
/snap/webots/14/usr/share/webots/webots: line 86: 260171 Segmentation fault      (core dumped) "$webots_home/bin/webots-bin" "$@"
```

DavidMansolino, 2020-04-29:
Did you try starting Webots with the safe mode enabled (https://cyberbotics.com/doc/guide/starting-webots#safe-mode)?

omichel, 2020-04-29:
Can you try this [fix](https://forum.snapcraft.io/t/snapped-app-not-loading-fonts-on-fedora-and-arch/12484/35) and report whether it helps?
```
sudo rm /var/cache/fontconfig/*
rm ~/.cache/fontconfig/*
fc-cache -r
```

Jajafarov, 2020-04-29:
Starting in safe mode does not help. The fontconfig cache fix didn't work either although it changed from `[40 210)` to `[0 205)` if that means anything.

Jajafarov, 2020-04-29:
Strange, clearing the cache without running `fc-cache -r` seems to work.

omichel, 2020-04-30:
Can you confirm what you did exactly? Simply:
```
sudo rm /var/cache/fontconfig/*
rm ~/.cache/fontconfig/*
```
And that worked?

Jajafarov, 2020-04-30:
> Can you confirm what you did exactly? Simply:
> 
> ```
> sudo rm /var/cache/fontconfig/*
> rm ~/.cache/fontconfig/*
> ```
> 
> And that worked?

Yes, I ran exactly this and it is now working.

PR #1618 (2020-05-04): Improve the snap documentation
Fixes #1601.
commit 435834aa (2020-05-04): Fixes #1601
closed 2020-05-04 by omichel