cyberbotics/webots #1602: detect whether or not executing from webots

LeoJhonSong, 2020-04-29:
**System**: Ubunutu18.04
**Version**: 2020b (Nightly Build (24-4-2020)

Hi, as I expect how my **real** robot behaves could be similar enough with how my **webots** robot behaves, I am trying to run both robot with same picec of code, which could detect whether to run a real robot or a webots robot by itself. So I tried to use the environment variables set by webots as a signal to show whether am I running my code from webots.

I added a line `os.system('export')` to my python3 controller **rover.py**, and followings are what I get:

- when I set the controller to **rover**, I got:
```shell
export WEBOTS_DISABLE_BINARY_COPY='True'
export WEBOTS_HOME='/usr/local/webots'
export WEBOTS_PIPE_IN='13'
export WEBOTS_ROBOT_ID='13'
export WEBOTS_SERVER='/tmp/webots_9322_667412782'
export WEBOTS_TMP_PATH='/tmp/webots-9316-8FRbqD/'
```

- when I set the controller to **extern** and run it from my VSCode, I got:
```shell
export WEBOTS_PIPE_IN='15'
```

Therefore my code frame goes like this:
```python
import os

flag_simulation = ('WEBOTS_PIPE_IN' in os.environ)

# if run for webots rover
if __name__ == "__main__" and flag_simulation:
    from controller import Robot
    # create the Robot instance.
    robot = Robot()
    # get the time step of the current world.
    timestep = int(robot.getBasicTimeStep())
    while robot.step(timestep) != -1:
        pass

# if run for real rover
if __name__ == "__main__" and not flag_simulation:
    while True:
        pass
```

However, supprisingly, I found that I cannot get the environment variable `WEBOTS_PIPPE_IN` with `os.environ`, which is wierd :drooling_face: with `print('WEBOTS_PIPE_IN' in os.environ)` in my code the output is **False**.

I wonder if I am doning the wrong way and is there any other way to detect whether I am running a webots robot? :joy: thx

DavidMansolino, 2020-04-29:
Using environment variables looks like a good idea, but why not simply checking for the ``WEBOTS_HOME`` one?
The problem with ``WEBOTS_PIPE_IN`` is that it is probably set during the creation of the Robot instance.

You may also simply add an argument to your code defining if you are using simulation or real robot.
Or even simpler, you might simply check if Webots is running. You can for example look for the process with something like (the drawback of this solution is that you have to make sure to start Webots **before** your controller):
```Python
import psutil    
"webots" in (p.name() for p in psutil.process_iter())
```

LeoJhonSong, 2020-04-29:
> ```python
> import psutil    
> "webots" in (p.name() for p in psutil.process_iter())
> ```

That could be reallly smart! Thanks a lot!

> but why not simply checking for the `WEBOTS_HOME` one?

The reason why I was trying to check for `WEBOTS_PIPE_IN` is that it seems there is not `WEBOTS_HOME` when run the webots robot from **extern**. As I always debug my robot, I run the webots robot form external more frequently.

Fortunately, your solution works prettry well for me. thx!

closed 2020-04-29 by LeoJhonSong