cyberbotics/webots #1603: Missing Device Fields Required for Automatic ROSification

lukicdarkoo, 2020-04-29:
**Is your feature request related to a problem? Please describe.**
In `webots_ros2` we would like to create automatic ROSification of the given robot (related PR https://github.com/cyberbotics/webots_ros2/pull/78). This means, that e.g.:
- All camera devices would be discovered, corresponding ROS2 topics of type `Image` and `CameraInfo` would be created, the camera would be enabled/disabled depending on the number of subscribers and transformation (`TransformStamped` messages with `tf2_ros`) of the camera in respect to the robot would be also published.
- All distance sensor device would be discovered, messages of type `Range` would be published, corresponding transforms as well.
- Similar applies for IMU, light sensors, LEDs and other devices.

However, we encountered a few obstacles:
- Cannot get a transformation from the robot (or world) to the device so we can publish `TransformStamped` messages.
- Cannot access to lookup table field of a distance sensor, light sensor or any other sensor, so we can get the actual value that we can publish as `Range`, `Illuminance` or similar.
- Cannot determine if InertialUnit, Gyro and Accelerometer are the part of the same IMU. This could be done by if knew that all the devices have the same transformation with respect to the robot. This assumption would not cover all the cases, but it would be configurable. 

All required fields (lookup table and transformations) are accessible if the robot is a supervisor. 

**Describe the solution you'd like**
Allow the robot to read the following fields:
- A **lookup table** of DistanceSensor, LightSensor and other sensors with a lookup table.
- **Transformation** of a device with respect to the robot/world origin.

This may not be exhaustive list of fields. Maybe we should finish the PR (https://github.com/cyberbotics/webots_ros2/pull/78) first assuming that we have access to all fields and then create an exhaustive list of fields that should be added to Webots.

**Describe alternatives you've considered**
Converting the robot to a supervisor robot was consider as an alternative. But, giving the role of supervisor to all robots that need to be ROSified destroys the purpose of the supervisor.

DavidMansolino, 2020-04-29:
This would imply to add these new functions (here in the DistanceSensor case):
```C
const double * wb_distance_sensor_get_translation(WbDeviceTag tag);  // return the translation of the device in the robot coordinate frame.
const double * wb_distance_sensor_get_rotation(WbDeviceTag tag);  // return the rotation of the device in the robot coordinate frame.
const double * wb_distance_sensor_get_lookup_table(WbDeviceTag tag);
int wb_distance_sensor_get_lookup_table_size(WbDeviceTag tag);
```

@lukicdarkoo let me know if I missed something.

lukicdarkoo, 2020-04-29:
Translation and rotation maybe should be common for all  [Devices](https://cyberbotics.com/doc/reference/device). Therefore, maybe:
```c
const double * wb_device_get_translation(WbDeviceTag tag);  // return the translation of the device in the robot coordinate frame.
const double * wb_device_get_rotation(WbDeviceTag tag);
```

lukicdarkoo, 2020-04-30:
Do you think if it is better just to read the actual value?
```c
double wb_distance_sensor_get_real_value(WbDeviceTag tag);
```
instead of
```c
const double * wb_distance_sensor_get_lookup_table(WbDeviceTag tag);
int wb_distance_sensor_get_lookup_table_size(WbDeviceTag tag);
```
This will reduce additional processing of the lookup table.

lukicdarkoo, 2020-04-30:
Oh, you are right, I completely forgot about it!

lukicdarkoo, 2020-08-07:
Implemented in #1706 and #1632

PR #1706 (2020-06-11): Allow Basic URDF Export
**Description**
According to the discussion, we should rather convert the node structure to URDF and publish transforms through `robot_state_publisher`. Therefore, this PR aims to allow export of stripped URDF version for `robot_state_publisher`. Extended, but still stripped version will be allowed later for RViz visualization.

URDF + `robot_state_publisher` vs publishing TF messages directly:
- Having URDF is much more flexible:
  - can be used to publish TF messages with `robot_state_publisher`,
  - can be used to visualize in RViz2,
  - can be used with MoveIt,
  -  [... cut]
closed 2020-08-07 by lukicdarkoo