cyberbotics/webots #1604: Can't change custom MFBoolField of a proto

GautierArcin, 2020-04-29:
Associated files (proto/world/controller) : 
[Bug report.zip](https://github.com/cyberbotics/webots/files/4552435/Bug.report.zip)

Can't change value of a custom MFBool field.

This field has 36 values. It's placed in the proto named `TileContainer`, which is a children of the proto `SpacecraftStructure`. 
The corresponding world is available in the archive and is named `FullMission.wbt`.

This bug has been seen on Webots R2020b, on comit  `fe5893b968e160447a52cc4619ad68a8f41f8022` compiled from the git.

**Steps to Reproduce**
1. Add the different proto and controller file to your project path
2. Open Webot and open `FullMission.wbt`
3. Make sure there is only one proto in the scene tree of type `SpacecraftStructure`, which has only one proto child of type `TileContainer`.
Make sure the proto child of type `TileContainer` has one supervisor named `controller_21`.
4. Launch the whole simulation.

**Expected behavior**
Supervisor should display the same output.

isLocked field should have different value in the scene tree. All values should be at `True` instead of `False`. 

**Screenshots**
The field "isLocked" doesn't change. 

[image]

Output of the supervisor: 
[image]
https://pastebin.com/bCN2Q9n0

**System**
 - Operating System: Ubuntu 16.04
 - Webots: Version R2020b, commit  `fe5893b968e160447a52cc4619ad68a8f41f8022`
 - Graphics Card: Nvidia gtx1050 4go
 - Processor: Intel(R) Core(TM) i5-6500 CPU

**Additional context**
Add any other context about the problem here.
Webots runned on Docker.
Docker version 19.03.5, build 633a0ea838

DavidMansolino, 2020-04-29:
Thank you for this bug report.
I had to slightly change the `TileContainer` PROTO as some dependency files are missing, but I can indeed reproduce the bug with a simple `TileContainer` PROTO file like this:
```
#VRML_SIM R2019b utf8
# license: Apache License 2.0
# license url: http://www.apache.org/licenses/LICENSE-2.0
# documentation url:
# A tile container.

PROTO TileContainer [
  field SFVec3f    translation            0 0 0                    # Is `Transform.translation`.
  field SFRotation rotation               0 1 0 0                  # Is `Transform.rotation`.
  field SFString   name                   "tile_container"         # Is `Solid.name`.
  field SFString   controller             "void"                   # Is `Robot.controller`.
  field SFString   controllerArgs         ""                       # Is `Robot.controllerArgs`.
  field SFBool     synchronization        TRUE                     # Is `Robot.synchronization`.
  field SFVec3f    translationBeam        -0.5 0 0                 # Is `Translation Beam`
  field MFBool     isLocked               [FALSE, FALSE, FALSE, FALSE, FALSE,
                                           FALSE, FALSE, FALSE, FALSE, FALSE,
                                           FALSE, FALSE, FALSE, FALSE, FALSE,
                                           FALSE, FALSE, FALSE, FALSE, FALSE,
                                           FALSE, FALSE, FALSE, FALSE, FALSE,
                                           FALSE, FALSE, FALSE, FALSE, FALSE,
                                           FALSE, FALSE, FALSE, FALSE, FALSE, FALSE] # isLocked field of the 36 hotdocks
]
{
  Robot {
    translation IS translation
    rotation IS rotation
    name IS name
    controller IS controller
    controllerArgs IS controllerArgs
    supervisor TRUE
    %{ local a = fields.isLocked.value }%
  }
}
```

The problem seems related to the fact that the node is in a field of another node. If I move the node to be a root node, then the behavior is correct (both in the controller and scene-tree).

We will investigate this bug and try to fix this as soon as possible (we will keep you posted about the progress here).

stefaniapedrazzi, 2020-06-11:
I tested on both on master and develop branch, but I can only reproduce it on the develop branch.
On the master branch, when I start the simulation, all the items in the `TileContainer.isLocked` field are correctly turned to TRUE.

stefaniapedrazzi, 2020-06-11:
The regression is due this change (introduced in #1331): https://github.com/cyberbotics/webots/blob/develop/src/lib/Controller/api/supervisor.c#L1502-L1515

On the master branch the DEF node is requested directly from Webots that seems to return the correct value.
On the develop branch, the DEF node is first searched in the internal node list where the node passed during the WbSupervisor `C_CONFIGURE` message is found. The `C_CONFIGURE` node (PROTO instance node) is not the same as the one returned by the `C_SUPERVISOR_NODE_GET_FROM_DEF` (PROTO parameter node). So when using it in the controller, the internal instantiated node is changes instead of the parameter node visible in the scene tree.

This is a very special case where the DEF node is also the "self" node of the controller.
I'm now checking if it is possible to adjust the node reference passed during the `C_CONFIGURE` message.

DavidMansolino, 2020-06-11:
Just for the note, this was an optimization so if it is causing issue (and we can't find any simple solution) we can remove this part without any problem (at least temporarily).

PR #1768 (2020-06-11): Fix self node in supervisor.c for a Robot proto parameter node
Fix #1604:
in case of PROTO and nested robots, the node passed during the supervisor `C_CONFIGURE` message is a PROTO parameter instance and not the node visible in the scene tree.
But the `wb_supervisor_node_get_from_def` function (and `wb_supervisor_node_get_self`) are supposed to return by default the visible nodes and not the instances, so that changes to the node are applied correctly from the Supervisor API in the exact same way they are applied from the scene tree.
closed 2020-06-29 by stefaniapedrazzi