cyberbotics/webots #1609: Python: Supervisor simulationReset() method doesn´t reset robot position

Dorteel, 2020-04-30:
**Describe the Bug**
I'm trying to use the simulationReset() method of the Supervisor, but it makes the  my world go crazy.  I'm supposed to record a sequence of motions as videos, then resetting the simulation at the end of the motion, changing some variables (the colour of the robot), and then recording again, multiple times. When resetting, the variables do not reset, and the inner loops that change the variables go nuts and out of sequence, and the robot does not get reset to the right location either.
The Supervisors simulationReset() method does not reset the robot position, only the robot´s joint angles.
In a simpler setting, I tried to use simulationReset() after playing a motion, when WeBots quit without a warning. In another setting I used motion.isOver() to run the simulationReset(), but then the simulation never resets.

**Steps to Reproduce**
The project is attached as a zip file
1. Open the world file
2. Run the controller: I commented out a lot of things to make the error a bit easier to see
3. The error occurs in the run() method.

**Expected behavior**
I expected the simulation to reset, just like when pressing the Simulation Reset button or Ctrl + Shift + F

**Attached project**
[Nao_1.zip](https://github.com/cyberbotics/webots/files/4559358/Nao_1.zip)

**System**

[image]
GeForce GTX 1650

DavidMansolino, 2020-05-01:
Thank you for this precise bug report.
However, I tried running your simulation and the only reset is happening at the first step (when the robot is already in the initial position). I let the simulation running for > 30min and no other reset happen, do we have to do something particular for this to happen?

DavidMansolino, 2020-06-04:
I am allowing myself to close this issue as we have no news about it.
Feel free to comment/re-open.

closed 2020-06-04 by DavidMansolino