cyberbotics/webots #1611: imporve experience of navigating with a touchpad

LeoJhonSong, 2020-04-30:
**System**: Ubuntu 1804
**Version**: 2020b

It seems webots's navigation gestures are mainly based on mouse events, while I am a big fan of touchpad and I do not have a mouse :cry: Recently I am struggling with **Rotate**, **Zoom** and **Alt F5**. I could not perform **Translate** or **Tilt** with my touchpad.

Could you please add some keybindings to do so? For example <kbd>Ctrl</kbd> and drag for **Translate**

DavidMansolino, 2020-05-01:
I confirm navigating in the 3D view using the touchpad is almost impossible.

omichel, 2020-05-01:
The key modifiers (Shift, Ctrl and Alt) are already used for moving objects, rotating them, applying a force or a torque. I believe the best way of improving the situation for touch screens would be to introduce multi-touch 3D navigation using [QtTouchEnvent](https://doc.qt.io/qt-5/qtouchevent.html) in addition to existing mouse events in [WbView3D.cpp](https://github.com/cyberbotics/webots/blob/master/src/webots/gui/WbView3D.cpp).

DavidMansolino, 2020-05-01:
> I believe the best way of improving the situation for touch screens would be 

Just for the note, this issue is not about touchscreens but touchpads.

omichel, 2020-05-01:
> Just for the note, this issue is not about touchscreens but touchpads.

Oh yes, you are right, I didn't noticed. However, the touch events of Qt also support track-pads, so it could be a nice way to address the problem.

LeoJhonSong, 2020-05-01:
@omichel It could be cool if touchscreen guestures are supported! (Although I don't have one)

Could you give some touch events example for **touchpad**? For me, almost all touchpad gesture with three or four fingers are already defined by my Ubuntu18.04. And there are even some gestures I could not use since my touchpad does not support :cry: 

Or could we do like drag with <kbd>t</kbd> for **translate**, scroll with <kbd>t</kbd> for **tilt**

Or even more wonderful, implement a set of **vim-like** navigation keybindings 😏

omichel, 2020-05-01:
I believe we could implement the same touchpad gestures as the touchscreen gestures already implemented in the web interface. Webots can probably override the gestures defined in the Ubuntu if we implement them in WbView3D.cpp.

omichel, 2020-05-01:
@LeoJhonSong: please go ahead with the implementation and let me know if you need help or have any further question. I will be happy to review your second PR 😄

LeoJhonSong, 2020-05-01:
I would like to give a try. The biggest problem I have is that I am totally unfamiliar with webots' source code. I am even not a good Cpp programer... Could you give me some keywords that I can search for, so that I can navigate to the code I need to focus to, in order to implement a vim-like navigation keybindings?
thx 😁

omichel, 2020-05-02:
Just [install the Webots development environment](https://github.com/cyberbotics/webots/wiki) and follow the hints I posted in my first message: you will have mainly to modify only [WbView3D.cpp](https://github.com/cyberbotics/webots/blob/master/src/webots/gui/WbView3D.cpp).

LeoJhonSong, 2020-05-02:
@omichel Thanks!

closed 2021-11-15 by BenjaminDeleze