cyberbotics/webots #1612: Improving Supervisor Sample

DavidMansolino, 2020-05-01:
**Describe the Bug**
We are getting questions about how to import and remove node in the scene almost every days (and sometimes even more than once per day).

We should improve our Supervisor sample (https://github.com/cyberbotics/webots/tree/master/projects/samples/devices/controllers/sample_supervisor) to add importation and removal of nodes.

ivy-2000, 2020-05-21:
Hi!
I'd like to work on this issue.
Could you give me a few pointers to start on it?
I understand that PROTO nodes are added directly by Add button. But how do we incorporate the importation of a node?

omichel, 2020-05-21:
Welcome. The idea would be to use the [wb_supervisor_field_import_*](https://cyberbotics.com/doc/reference/supervisor#wb_supervisor_field_import_mf_node) functions to import a new node in a world file while the simulation is running. Later on, we should also use [wb_supervisor_node_remove](https://cyberbotics.com/doc/reference/supervisor#wb_supervisor_node_remove) to remove a node also while the simulation is running. The imported node may be a PROTO, but not necessarily.

closed 2020-07-14 by DavidMansolino