cyberbotics/webots #1613: Streaming server crash

omichel, 2020-05-01:
Running [ac3_viewer.wbt](https://github.com/cyberbotics/webots/blob/master/projects/humans/c3d/worlds/c3d_viewer.wbt) with streaming crashes Webots on Linux.

Step to reproduce:
1. Start Webots in streaming mode with [ac3_viewer.wbt](https://github.com/cyberbotics/webots/blob/master/projects/humans/c3d/worlds/c3d_viewer.wbt) with `--mode=pause`.
2. Open the [streaming viewer page](https://github.com/cyberbotics/webots/blob/master/resources/web/streaming_viewer/index.html) and connect to Webots.
3. Press the `Run` button in Webots.
4. Webots crashes.

Stack:
```
#0  0x00007fdc16e62e97 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007fdc16e64801 in __GI_abort () at abort.c:79
#2  0x00007fdc17857957 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007fdc1785dae6 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007fdc1785db21 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007fdc1785dd54 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007fdc17859837 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x000055648a64beff in std::__detail::_Map_base<WbTriangleMeshCache::IndexedFaceSetKey, std::pair<WbTriangleMeshCache::IndexedFaceSetKey const, WbTriangleMeshCache::TriangleMeshInfo>, std::allocator<std::pair<WbTriangleMeshCache::IndexedFaceSetKey const, WbTriangleMeshCache::TriangleMeshInfo> >, std::__detail::_Select1st, std::equal_to<WbTriangleMeshCache::IndexedFaceSetKey>, WbTriangleMeshCache::IndexedFaceSetKeyHasher, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true>, true>::at(WbTriangleMeshCache::IndexedFaceSetKey const&) (this=0x55648ace8a00 <WbIndexedFaceSet::cTriangleMeshMap>, __k=...)
    at /usr/include/c++/7/bits/hashtable_policy.h:774
#8  0x000055648a64baf1 in std::unordered_map<WbTriangleMeshCache::IndexedFaceSetKey, WbTriangleMeshCache::TriangleMeshInfo, WbTriangleMeshCache::IndexedFaceSetKeyHasher, std::equal_to<WbTriangleMeshCache::IndexedFaceSetKey>, std::allocator<std::pair<WbTriangleMeshCache::IndexedFaceSetKey const, WbTriangleMeshCache::TriangleMeshInfo> > >::at(WbTriangleMeshCache::IndexedFaceSetKey const&) (this=0x55648ace8a00 <WbIndexedFaceSet::cTriangleMeshMap>, __k=...)
    at /usr/include/c++/7/bits/unordered_map.h:990
#9  0x000055648a649c05 in WbIndexedFaceSet::exportNodeHeader(WbVrmlWriter&) const (this=0x5564909fc7f0, writer=...) at nodes/WbIndexedFaceSet.cpp:705
#10 0x000055648a6b51ec in WbNode::writeExport(WbVrmlWriter&) const (this=0x5564909fc7f0, writer=...) at vrml/WbNode.cpp:1098
#11 0x000055648a6b4194 in WbNode::write(WbVrmlWriter&) const (this=0x5564909fc7f0, writer=...) at vrml/WbNode.cpp:959
#12 0x000055648a7420df in WbSFNode::write(WbVrmlWriter&) const (this=0x55649dfc4bd0, writer=...) at vrml/WbSFNode.cpp:57
#13 0x000055648a6b5004 in WbNode::exportNodeSubNodes(WbVrmlWriter&) const (this=0x55649dfba390, writer=...) at vrml/WbNode.cpp:1073
#14 0x000055648a6b51b9 in WbNode::exportNodeContents(WbVrmlWriter&) const (this=0x55649dfba390, writer=...) at vrml/WbNode.cpp:1094
#15 0x000055648a6b5210 in WbNode::writeExport(WbVrmlWriter&) const (this=0x55649dfba390, writer=...) at vrml/WbNode.cpp:1100
#16 0x000055648a6b4194 in WbNode::write(WbVrmlWriter&) const (this=0x55649dfba390, writer=...) at vrml/WbNode.cpp:959
#17 0x000055648a694a3a in WbMFNode::writeItem(WbVrmlWriter&, int) const (this=0x556490cc3420, writer=..., index=0) at vrml/WbMFNode.cpp:193
#18 0x000055648a694c0d in WbMFNode::write(WbVrmlWriter&) const (this=0x556490cc3420, writer=...) at vrml/WbMFNode.cpp:215
#19 0x000055648a6b5004 in WbNode::exportNodeSubNodes(WbVrmlWriter&) const (this=0x55649dfc4c70, writer=...) at vrml/WbNode.cpp:1073
#20 0x000055648a6b51b9 in WbNode::exportNodeContents(WbVrmlWriter&) const (this=0x55649dfc4c70, writer=...) at vrml/WbNode.cpp:1094
#21 0x000055648a6b5210 in WbNode::writeExport(WbVrmlWriter&) const (this=0x55649dfc4c70, writer=...) at vrml/WbNode.cpp:1100
#22 0x000055648a6b4194 in WbNode::write(WbVrmlWriter&) const (this=0x55649dfc4c70, writer=...) at vrml/WbNode.cpp:959
#23 0x000055648a797c91 in WbStreamingServer::propagateNodeAddition(WbNode*) (this=0x55648b1013b0, node=0x55649dfc4c70) at gui/WbStreamingServer.cpp:770
#24 0x000055648a79c078 in QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<WbNode*>, void, void (WbStreamingServer::*)(WbNode*)>::call(void (WbStreamingServer::*)(WbNode*), WbStreamingServer*, void**) (f=
    (void (WbStreamingServer::*)(WbStreamingServer * const, WbNode *)) 0x55648a797934 <WbStreamingServer::propagateNodeAddition(WbNode*)>, o=0x55648b1013b0, arg=0x7ffd65986c90) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:152
#25 0x000055648a79bbc4 in QtPrivate::FunctionPointer<void (WbStreamingServer::*)(WbNode*)>::call<QtPrivate::List<WbNode*>, void>(void (WbStreamingServer::*)(WbNode*), WbStreamingServer*, void**) (f=
    (void (WbStreamingServer::*)(WbStreamingServer * const, WbNode *)) 0x55648a797934 <WbStreamingServer::propagateNodeAddition(WbNode*)>, o=0x55648b1013b0, arg=0x7ffd65986c90) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:185
#26 0x000055648a79b267 in QtPrivate::QSlotObject<void (WbStreamingServer::*)(WbNode*), QtPrivate::List<WbNode*>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=1, this_=0x55648b04ea30, r=0x55648b1013b0, a=0x7ffd65986c90, ret=0x0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:414
#27 0x00007fdc1c021736 in QMetaObject::activate(QObject*, int, int, void**) () at /home/cyberbotics/webots/lib/webots/libQt5Core.so.5
#28 0x000055648a8cba6c in WbNodeOperations::nodeAdded(WbNode*) (this=0x55648b04e650, _t1=0x55649dfc4c70) at build/debug/WbNodeOperations.moc.cpp:164
#29 0x000055648a6c1100 in WbNodeOperations::importNode(WbNode*, WbField*, int, QString const&, QString const&, bool, int*, bool) (this=0x55648b04e650, parentNode=0x556490c445c0, field=0x556490de5fa0, itemIndex=0, filename=..., nodeString=..., avoidIntersections=false, importedNodesNumber=0x7ffd659871b0, fromSupervisor=true) at nodes/utils/WbNodeOperations.cpp:185
#30 0x000055648a6c0921 in WbNodeOperations::importNode(int, int, int, QString const&, QString const&, int*, bool) (this=0x55648b04e650, nodeId=155, fieldId=4, itemIndex=0, filename=..., nodeString=..., importedNodesNumber=0x7ffd659871b0, fromSupervisor=true) at nodes/utils/WbNodeOperations.cpp:107
#31 0x000055648a7a128a in WbSupervisorUtilities::handleMessage(QDataStream&) (this=0x556490ad6170, stream=...) at nodes/utils/WbSupervisorUtilities.cpp:1104
#32 0x000055648a724c4d in WbRobot::handleMessage(QDataStream&) (this=0x556490c445c0, stream=...) at nodes/WbRobot.cpp:970
#33 0x000055648a723c3c in WbRobot::dispatchMessage(QDataStream&) (this=0x556490c445c0, stream=...) at nodes/WbRobot.cpp:753
#34 0x000055648a5da9da in WbController::readRequest() (this=0x55649cb8e360) at control/WbController.cpp:1070
#35 0x000055648a8ba8aa in WbController::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) (_o=0x55649cb8e360, _c=QMetaObject::InvokeMetaMethod, _id=2, _a=0x7ffd659875e0) at build/debug/WbController.moc.cpp:147
#36 0x00007fdc1c021411 in QMetaObject::activate(QObject*, int, int, void**) () at /home/cyberbotics/webots/lib/webots/libQt5Core.so.5
#37 0x00007fdc1c021411 in QMetaObject::activate(QObject*, int, int, void**) () at /home/cyberbotics/webots/lib/webots/libQt5Core.so.5
#38 0x00007fdc1baa8843 in QAbstractSocketPrivate::emitReadyRead(int) () at /home/cyberbotics/webots/lib/webots/libQt5Network.so.5
#39 0x00007fdc1baa88fc in QAbstractSocketPrivate::canReadNotification() () at /home/cyberbotics/webots/lib/webots/libQt5Network.so.5
#40 0x00007fdc1babad11 in QReadNotifier::event(QEvent*) () at /home/cyberbotics/webots/lib/webots/libQt5Network.so.5
#41 0x00007fdc1a5fff6c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /home/cyberbotics/webots/lib/webots/libQt5Widgets.so.5
#42 0x00007fdc1a607120 in QApplication::notify(QObject*, QEvent*) () at /home/cyberbotics/webots/lib/webots/libQt5Widgets.so.5
#43 0x00007fdc1bff46c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /home/cyberbotics/webots/lib/webots/libQt5Core.so.5
#44 0x00007fdc1c04d5b8 in socketNotifierSourceDispatch(_GSource*, int (*)(void*), void*) () at /home/cyberbotics/webots/lib/webots/libQt5Core.so.5
#45 0x00007fdc142e7417 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
[... cut ...]
#48 0x00007fdc1c04c97f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/cyberbotics/webots/lib/webots/libQt5Core.so.5
#49 0x00007fdc1bff3082 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/cyberbotics/webots/lib/webots/libQt5Core.so.5
#50 0x00007fdc1bffba80 in QCoreApplication::exec() () at /home/cyberbotics/webots/lib/webots/libQt5Core.so.5
#51 0x000055648a62bafa in WbGuiApplication::exec() (this=0x7ffd65987d40) at gui/WbGuiApplication.cpp:262
#52 0x000055648a8238dd in main(int, char**) (argc=6, argv=0x7ffd65987e98) at gui/main.cpp:207
```

omichel, 2020-05-01:
A few notes:
- the crash doesn't occur on Windows.
- it crashes because of the insertion of an object in the scene from the controller [here](https://github.com/cyberbotics/webots/blob/master/projects/humans/c3d/controllers/c3d_viewer/c3d_viewer.py#L192-L193).
- it seems two objects are created (one for which the Wren finalization is called and a ghost object for which no Wren finalization is called). Then Webots tries to send an update to `webots.js` of the ghost object and crashes [here](https://github.com/cyberbotics/webots/blob/master/src/webots/nodes/WbIndexedFaceSet.cpp#L705) because the `mMeshKey` is 0 as the object was not finalized.
- I don't understand why we need to create this ghost object and why we are trying to send the update to `webots.js` on this ghost object.

omichel, 2020-05-01:
I wrote a simplified example exhibiting the problem.

simple.wbt
```
#VRML_SIM R2020a utf8
WorldInfo {
  title "C3D viewer"
}
Viewpoint {
  orientation -0.7095662703664664 0.642643640090935 0.2890066777929114 0.7904657261701156
  position 3.271213250962302 4.245847478516282 5.147248384110216
}
TexturedBackground {
}
TexturedBackgroundLight {
}

RectangleArena {
  floorSize 10 2
  wallHeight 0.001
}

Supervisor {
  controller "simple"
}
```

simple.py:
```python
from controller import Supervisor

supervisor = Supervisor()
children = supervisor.getSelf().getField('children')
shape = """DEF SHAPE Shape {
appearance PBRAppearance {
baseColor 0.264 0.264 0.8
roughness 0.19999999999999996
metalness 0
}
geometry IndexedFaceSet {
coord Coordinate {
  point [
    -0.2 0 0.2
    0.2 0 0.2
    0.2 0 -0.2
    -0.2 0 -0.2
    -0.1 0.1 -0.1
    -0.1 0.1 0.1
    0.1 0.1 0.1
    0.1 0.1 -0.1
  ]
}
coordIndex [
  0, 1, 6, 5, -1, 2, 7, 6, 1, -1, 3, 4, 7, 2, -1, 0, 5, 4, 3, -1, 7, 4, 5, 6, -1, 3, 2, 1, 0, -1
]
}
}"""
children.importMFNodeFromString(-1, shape)
while supervisor.step(32) != -1:
    pass
```

If you replace:
```
Supervisor {
  controller "simple"
}
```

with:

```
Robot {
  supervisor TRUE
  controller "simple"
}
```

Then, it doesn't crash any more...
So, it crashes only if the parent node of the inserted node is a PROTO (`Supervisor` is a deprecated PROTO).

stefaniapedrazzi, 2020-05-01:
The "ghost" object is the node in the PROTO parameter. Actually it is simply a non-finalized node that is needed to forward the changes to the instances used inside the PROTO node.
So it is correct that two nodes are created when inserting the node from the controller: one is the non-finalized instance in the PROTO parameter and one is the finalized node instance in the PROTO body.

But the PROTO parameter node should not be sent to `webots.js`.

omichel, 2020-05-01:
OK, thank you for the explanations. I believe I found a fix. I will write a PR for that and ask you to review it as I am not sure it's the right way to fix the problem.

PR #1614 (2020-05-04): Fix streaming crash
Attempt to fix #1613.
closed 2020-05-04 by omichel