cyberbotics/webots #1615: Webots Floor vibrates with the robot

tahir1069, 2020-05-01:
**Floor vibrates when the simulation is started**
The warthog model in Webots when launched the floor starts to vibrate.

**Steps to Reproduce**
1. Launch the world warthog_dev_no_assembly.wbt.
2. Run controller(from mbs_communicator package) in a separate terminal.
3. In another terminal run mbs_communicator_script(from mbs_communicator package).
3. This should start moving the robot.
4. The robot moves and the floor vibrates with it.

**Expected behavior**
Smooth movement of robot.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: Linux Ubuntu 18.04

[warthog_issue.zip](https://github.com/cyberbotics/webots/files/4565233/warthog_issue.zip)

tahir1069, 2020-05-01:
Actually this was a problem with my viewpoint in Webots which I changed and the problem was solved. So I think this actually is not an issue so if the admins want to delete this please do so.

closed 2020-05-01 by tahir1069

DavidMansolino, 2020-05-02:
Nice to see that you found the cause of the problem, thank you for the feedback!