cyberbotics/webots #1619: X3D texture path conflict

omichel, 2020-05-04:
# Description

Textures displayed in the X3D file or web interface may be wrong.

# Conditions

- A proto file uses a texture image named `texture.jpg` which is located in its `protos/textures` folder.
- A world file uses the above proto file but also a different texture image named `texture.jpg` in its `worlds/textures` folder.

# Consequence

The X3D model or web interface will display the wrong texture either for the world or for the proto.

# Explanation

This is due to the fact that the proto and world textures in X3D files are all added inside a unique `textures` folder.

# Proposed solution

The texture url in the X3D files should refer to world and proto textures in this way: `worlds/textures/texture.jpg` and `protos/textures/texture.jpg` instead of `textures/texture.jpg`.

stefaniapedrazzi, 2020-05-04:
This is a known limitation of textures export.
Note that your proposal only fixes a special case but not all the issues with texture naming conflicts.
For example if two different PROTO nodes uses two different textures with the same name `texture.jpg`, then the appearance of the second PROTO will be wrong. This could happen is at least one PROTO is a default one.

omichel, 2020-05-04:
The case you mention refers to a PROTO is not inside the local project folder, but somewhere in the Webots global `projects` folder. In such a case, if we can guarantee the Webots global `projects` folder has unique proto texture names, then, there should be no possible conflict for users.
Because in case the local `protos` folder contains subfolders, these subfolders should be part of the texture url, e.g., `protos/subfolder/textures/texture.jpg`.
This cover all the cases, isn't it?

stefaniapedrazzi, 2020-05-04:
Yes.
In order to work, also Webots and project PROTO textures have to be stored in different folders.
And texture names for Webots PROTOs should be unique or you have to store each texture in a separate folder specific to the PROTO (unique PROTO name).

omichel, 2020-05-04:
Yes. And if we want to get rid of the uniqueness constraint for Webots PROTO textures, we should introduce a new type of url, like `webots/projects/devices/microsoft/protos/textures/kinect.jpg` which would refer to a Webots PROTO texture instead of a local texture (with the url starting with `worlds/` or `protos/`. This is probably the best solution as it will save us from ensuring the uniqueness of textures names in Webots PROTOs.

DavidMansolino, 2020-05-05:
> The case you mention refers to a PROTO that is not inside the local project folder, but somewhere in the Webots global `projects` folder. In such a case, if we can guarantee the Webots global `projects` folder has unique proto texture names, then, there should be no possible conflict for users.
> Because in case the local `protos` folder contains subfolders, these subfolders should be part of the texture url, e.g., `protos/subfolder/textures/texture.jpg`.

Are you sure? To me it looks like this structure is possible:
```
my_project
    -> protos
       -> sub_folder_A
           -> A.proto
           -> textures
                -> my_texture.jpg
       -> sub_folder_B
           -> B.proto
           -> textures
                -> my_texture.jpg
```

omichel, 2020-05-05:
> Are you sure? To me it looks like this structure is possible:

Yes, and in that case the texture urls should be `protos/sub_folder_A/textures/my_texture.jpg` for the first one and `protos/sub_folder_B/textures/my_texture.jpg` for the second one. So the problem is solved by this approach.

DavidMansolino, 2020-05-05:
Currently, we do support to have this `textures/my_texture.jpg` even for sub-folders protos in the local folder (and the correct texture is used in my previous example for example), so having to put the complete path is a huge regression IMO because it will make the protos to be location dependent and it will be a pain to have to update all the textures paths when changing its location.

DavidMansolino, 2020-05-05:
We had a misunderstanding about how to fix this, we clarified this by phone.
The texture url will be changed only in the X3D export and not in Webots (I originally thought that both will be changed).

omichel, 2020-05-06:
Stepping back from this issue, it seems my proposal is like putting a dirty patch on the problem.
I believe we should take inspiration from the original VRML97 design and in particular the concept of URL to fix this problem in a more generic and cleaner way. I will come up with a hopefully better solution. Meanwhile, I am closing this issue and there is no point fixing it if we introduce a better URL system.

closed 2020-05-06 by omichel