cyberbotics/webots #1625: Error while usingMatlab with WEBOTS

Mohammad9966, 2020-05-04:
I'm trying to control an E-puck robot using Matlab but it's still given me the following error in Matlab :

> "

 Warning: Name is nonexistent or not a directory: C:\Webots\lib\matlab 
> In path (line 109)
  In add path (line 88)
  In launcher (line 23) 
Undefined variable "Matlab" or class "matlab.addons.installedAddons".

Error in launcher (line 28)
  installed_addons = 
 matlab.addons.installedAddons;"

what should I do?

 
»

DavidMansolino, 2020-05-05:
Which version of Webots are you using and on which OS?

Mohammad9966, 2020-05-05:
2020, the latest one and im using win 10

omichel, 2020-05-05:
I would recommend you to use the latest nightly build of Webots R2020a-rev2 from [here](https://github.com/cyberbotics/webots/releases/tag/nightly_4_5_2020) as we fixed some issues with respect to MATLAB since the release of Webots R2020-rev1.
From the error message you get, it seems you installed Webots for all users in the `C:\Webots` folder. Is that correct?
By the way, which version of MATLAB are you using?
Are you trying to run the [e-puck_matlab](https://github.com/cyberbotics/webots/blob/master/projects/languages/matlab/worlds/e-puck_matlab.wbt) demo world?

Mohammad9966, 2020-05-05:
Im using matlab 2017 , no im just testing the code on the website under tutorial sectio
I will try and tell you , pls keep in touch these days im away off beyond the dead line to my project deadline and i need help
Thx

Mohammad9966, 2020-05-05:
I should download the .exe file only?

omichel, 2020-05-05:
Yes, [this one](https://github.com/cyberbotics/webots/releases/download/nightly_4_5_2020/webots-R2020a-rev2_setup.exe).

Mohammad9966, 2020-05-07:
@omichel 
man thank u so much, it was really helpful installing this version, after 2 weeks of suffering, finally, I get out from this error but now I got another one but I think it's simple 

[my_controller12] Error: The specified procedure could not be found.
[my_controller12]  (dynamic library)
[my_controller12] Error: remote control initialisation failed

omichel, 2020-05-07:
Yes, that one should be easier to fix.

1. Can you open [nao_matab.wbt](https://github.com/cyberbotics/webots/blob/master/projects/robots/softbank/nao/worlds/nao_matlab.wbt) file from `Files` &rarr; `Open Samples Worlds...` &rarr; `robots` &rarr; `softbank` &rarr; `nao` &rarr; `nao_matlab.wbt` and run it?
2. Can you open [e-puck_matlab.wbt](https://github.com/cyberbotics/webots/blob/master/projects/languages/matlab/worlds/e-puck_matlab.wbt) from `Files` &rarr; `Open Samples Worlds...` &rarr; `languages` &rarr; `matlab` &rarr; `nao_matlab.wbt` and run it?

Does your world file uses an e-puck robot?

Mohammad9966, 2020-05-09:
Yes, both of them worked and I'm using an e-puck robot

omichel, 2020-05-11:
This is good news that e-puck_matlab.wbt is working for you. That means you MATLAB and Webots setup is OK. Now the problem is specific to your simulation. Did you modify the e-puck robot in your simulation? Especially, did you changed its remote control library, or maybe you copied it in your local folder and didn't compiled it? Can you check if your local project has a `plugins` folder which contains a `remote_controls` subfolder? If so, simply delete the `remote_control` subfolder and restart Webots with your simulation. Let me know if that resolves the problem.

Mohammad9966, 2020-05-12:
I have received the following :
[matlab] Error: The specified procedure could not be found.
[matlab]  (dynamic library)
[matlab] Error: remote control initialisation failed
[matlab] Error: Cannot load the "C:/Webots/projects/robots/gctronic/e-puck/plugins/remote_controls/e-puck_bluetooth/e-puck_bluetooth.dll" remote control library.
[matlab] Running Matlab sample Webots controller.

and the robot run until it hits a purple square 
then I received the following Error:
WARNING: DEF E_PUCK E-puck (PROTO) > HingeJoint > RotationalMotor: The requested velocity -7.13803 exceeds 'maxVelocity' = 6.28.
WARNING: DEF E_PUCK E-puck (PROTO) > HingeJoint > RotationalMotor: The requested velocity -7.02126 exceeds 'maxVelocity' = 6.28.
WARNING: DEF E_PUCK E-puck (PROTO) > HingeJoint > RotationalMotor: The requested velocity -9.36128 exceeds 'maxVelocity' = 6.28.

and when i try to change the speed he asked me to transfer the necessary files from the source to the target , when i select another directory , it says : 
Target directory is not empty.
Please select another directory.
if select another one will it affect the Matlab tracking the bath?

Mohammad9966, 2020-05-12:
Also, can I add an Omni-wheel robot to the environment?

omichel, 2020-05-12:
I would recommend you then to restart from the *e-puck_matlab.wbt* example, slightly modify the MATLAB source code from the Webots text editor, save it so that Webots will propose you to re-locate the project in your own folder. Then, from there, you will be able to continue to modify the example until it matches your needs. And of course, yes, you can add an omni-wheel robot if needed.

Mohammad9966, 2020-05-14:
ok, is there an Omni-wheel robot in the simulator itself or should I build my own one and import it to the simulator?

omichel, 2020-05-14:
Yes: https://cyberbotics.com/doc/guide/samples-howto#omni_wheels-wbt

closed 2020-05-14 by omichel