cyberbotics/webots #1634: Reflection intensity of Lidar

jrcblue, 2020-05-05:
Hi, I was wondering whether lidar in Webots can get the reflection intensity, which is useful in some 3D object detection algorithm.
Thanks!

liuton95, 2020-05-05:
Yes, like KITTI dataset, in their lidar raw data, each point is stored with its (x, y, z) coordinate and an additional reflectance value (r)

DavidMansolino, 2020-05-06:
This is indeed a limitation of the current model of the Lidar in Webots now, it does not simulate intensity, but only distance/position.

closed 2020-05-06 by DavidMansolino

mwopfner, 2021-11-18:
Is there a plan/timeline to implement this feature?

omichel, 2021-11-19:
No, unless someone wants to fund this implementation.

cosmicog, 2021-12-06:
In my opinion, without the intensity value, we can only simulate small range of dud lidars which are more suitable for hobby projects, not professional robotics applications.

Could you maybe give us some pointers on where to start implementing this, so I (or someone else) might want to implement and contribute.

omichel, 2021-12-06:
> Could you maybe give us some pointers on where to start implementing this, so I (or someone else) might want to implement and contribute.

The main source code for the Lidar node is in [WbLidar.cpp](https://github.com/cyberbotics/webots/blob/master/src/webots/nodes/WbLidar.cpp), but it also involves many other classes as you will see. The lidar depth values are computed in an OpenGL shader based on the OpenGL depth buffer. In a first approach, the reflection value could be computed in a shader as well, based on the OpenGL color buffer. Then we could refine it based on some PBR appearance computed from the appearance and PBR texture of the object. It is important that this computation is performed in a shader to be efficient.

ZiqiTIAN, 2022-05-25:
Hi, 

> In my opinion, without the intensity value, we can only simulate small range of dud lidars which are more suitable for hobby projects, not professional robotics applications.
>
> Could you maybe give [...]

Hi, did you finally solve this problem?

cosmicog, 2022-05-25:
@ZiqiTIAN Hello, Unfortunately it is still in my TODO list. Right now I'm mostly working on a robot that doesn't need intensity values.

nazmicancalik, 2023-02-27:
Hey, I am also interested in this feature. Any developments on this so far?