cyberbotics/webots #1637: rostopic echo /......./...../recognition_objects doesnt work error. webots

Mathiebhan, 2020-05-06:
**Describe the Bug**
Trying to get the topic to work so i can get object position/orientation using object recognition. Which the topic 
/<device_name>/recognition_objects | topic | webots_ros::RecognitionObject
doesnt work. 
Maybe you guys have a solution for it :)

**Screenshots**
I launch my ros node. Then i try to echo the ros topic which gives error as seen in red below in the picture. 
[image]

**System**
 - Operating System: Tried on Linux Ubuntu 18.04 ros Melodic and ros kinetic 16.04] same issue.

DavidMansolino, 2020-05-08:
I can indeed reproduce this issue, it seems the md5 sum is wrong on the Webots side.

DavidMansolino, 2020-05-08:
This has been fixed in #1641, the patch will be available in the next release of Webots.
In the meantime, you can use the nightly builds from tomorrow available here: https://github.com/cyberbotics/webots/releases

closed 2020-05-08 by DavidMansolino