cyberbotics/webots #1638: Better URL support in wbt files

omichel, 2020-05-06:
# Problem with the current design

It is currently pretty complicated to understand which textures, protos, robot controllers, sounds or plugin are used by looking only at the content of a *wbt* file. This is because there are many search paths with [complex priority rules](https://www.cyberbotics.com/doc/reference/imagetexture#search-rule-of-the-texture-path). This defeats the principle of Uniform Resource Locator (URL).

This also yields to problems when exporting to X3D format for example, where it is difficult to refer to the correct resources used by the X3D file (see, [textures name clash](#1619)).

# Proposed solution

We should take inspiration from the original VRML97 design principles where resources are located very precisely thanks to a better URL implementation.

1. The URL fields should be implemented as real URLs, either absolute or relative to the *wbt*/*proto* file:
  - `ImageTexture.url`
  - `Mesh.url`
  - `Background.*Url`
  - `Camera.noiseMaskUrl`
  - `ContactProperties.bumpSound`
  - `ContactProperties.rollSound`
  - `ContactProperties.slideSound`
  - `Skin.model`
  - `Robot.window [code]`
  - `Robot.controller [code]`
  - `Robot.remoteControl [code]`
  - `WorldInfo.physics [code]`
Examples:
  - `url "textures/image.jpg"`
  - `url "https://webots.cloud/R2021a/textures/webots_billboard.jpg"`
  - `url "../textures/protos/textures/card.jpg"`

2. Each PROTO used in a *wbt*/*proto* file should be declared at the beginning of the file using the simplified **EXTERNPROTO** syntax:
- `EXTERNPROTO "../protos/MyProto.proto"`
- `EXTERNPROTO "https://webots.cloud/R2021a/robots/boston_dynamics/spot/protos/Spot.proto"`
- `EXTERNPROTO "https://mywebsite.com/myfolder/MyBot.proto"`
All the resources (images and subsequent **EXTERNPROTO** will be downloaded as cascading URLs).
Icon URLs will be forged automatically by Webots to download the corresponding PROTO icons.

3. For URLs referring to code (marked `[code]` in the above list), one should consider they refer to the folder containing the code to be used by Webots or to the file to be used by Webots. Examples:
- `controller "../controllers/my_controller/"` will launch `my_controller[.exe]` located inside the `my_controller` folder.
- `controller "../controllers/my_controller/my_controller.py"` will launch `my_controller.py` with the default Python interpreter.
- `"../plugins/physics/my_physics/"` will launch `my_physics.dll` (or `.dylib` or `.so` depending on the platform).
- `"../plugins/robot_windows/my_robot_window/"` will use `my_robot_window.html` and `my_robot_window.js` from this folder
All the dependencies (if any) of the should be in the same folder as the code URL.
In case of a local URL, nothing need to be done.
In case of a remote URL for a controller, remote control or physics plugin, Webots should download a ZIP file containing the whole folder: `"controller "https://mysite.com/my_controller/my_controller.py"` &rarr; Webots will download `"https://mysite.com/my_controller.zip"` unzip it, store in its cache and use it.

Webots will implement a cache for these resources so that for example `https://webots.cloud/R2021a/robots/boston_dynamics/spot/protos/Spot.proto` and its subsequent resources are downloaded only once and stored for subsequent use in WEBOTS_HOME/cache/https/webots.cloud/R2021a/robots/boston_dynamics/spot/protos/Spot.proto`, etc.

# Benefits of the proposed solution

We will be able to remove any constraint on the webots project directory structure: a world file could be anywhere, including at root level and it will refer to its resources in a very explicit way. Such resources could be organized in any directory structure at the users' choice.

Webots will be more efficient at loading the resources of world files, as it will not have to explore a priority ordered list of folders, but it will load directly the file explicitly referred to by the *wbt*/*proto* file. Such a list may be very long in case of protos resources for example.

It will be simpler for the user to understand which resource file is actually loaded by Webots.

We won't need to distribute the Webots resources (textures, protos, controllers, plugins, etc.) in the Webots installation package. Instead, they will be downloaded automatically as the user needs them. This should make the installation package much lighter.

It will be convenient to use resources created by others using **EXTERNPROTO** with absolute URLs for example.

It will be easier to deploy Webots simulations in the cloud.

LeoJhonSong, 2020-05-19:
That would be really cool!

omichel, 2022-06-13:
1, 2 and 4 are implemented in Webots R2022b.

BenjaminDeleze, 2023-06-13:
Do we want to implement point 3 in the future or can we close  this?

omichel, 2023-06-13:
Yes, I believe that implementing point 3 would make the world file format more consistent. So, let's keep it open.