cyberbotics/webots #1642: Invalid MATLAB Controller Function Prototypes for Arrays

lukicdarkoo, 2020-05-08:
**Documentation page(s) concerned**
- https://cyberbotics.com/doc/guide/controller-programming?tab-language=matlab#reading-sensors
- https://cyberbotics.com/doc/reference/accelerometer?tab-language=matlab#accelerometer-functions
- https://cyberbotics.com/doc/reference/compass?tab-language=matlab#compass-functions
- ...

**What is the problem with the current documentation?**
As discussed [here](https://github.com/cyberbotics/webots/pull/1632#discussion_r422013099) examples such as:
```
[x y z] = wb_accelerometer_get_values(acc)
```
are not valid in MATLAB and get be misleading for the users.

**Describe how you would fix it**
Those examples should be replaced with
```
x_y_z_array = wb_accelerometer_get_values(acc)
```

gokulp01, 2020-05-14:
@omichel @lukicdarkoo I think #1664 solves the issue :)

omichel, 2020-05-18:
@lukicdarkoo: can we close this issue?

lukicdarkoo, 2020-05-18:
Yes!

closed 2020-05-18 by lukicdarkoo