cyberbotics/webots #1650: Errors in Windows Package Creation

DavidMansolino, 2020-05-10:
**Describe the Bug**
The creation of the Windows package is generating error warning messages:
```
# /mingw64/bin/libzstd.dll is already included
# processing ffmpeg dependencies (DLLs)
# building distribution source
# creating webots.iss version R2020a revision 2: done
# creating webots_setup.exe (takes long)
Error on line 21437 in C:\projects\webots-master\src\packaging\webots.iss: Source file "C:\msys64\\mingw64\bin\libicuuc65.dll" does not exist.
Compile aborted.
```

https://ci.appveyor.com/project/omichel/webots-master/builds/32774904#L5582

omichel, 2020-05-10:
Yes, this is because libicu was upgraded to version 66. I will take care of fixing this.

omichel, 2020-05-11:
The package now builds without any error or warning: https://ci.appveyor.com/project/omichel/webots-9ufow

PR #1651 (2020-05-11): Fixed Windows packaging
Fix #1650.
I am setting a generic version name for this *libicu* library whose version is changing quite often, so that this bug shouldn't strike back anytime soon.
closed 2020-05-11 by omichel