cyberbotics/webots #1652: Telemetry data and GDPR

stefaniapedrazzi, 2020-05-11:
If telemetry data is sensitive data, we should make Webots compliant to GDPR we might adapt the first launch window:
* the user should actively check/uncheck the telemetry check box
* add a Privacy Policy
* give the possibility to users to delete data collected about them

It is probably arguable if telemetry data is sensitive. But given to this description we are one the grey area:
> Do note that according to GDPR, combinations of information that may not be unique in isolation can potentially identify an Individual. So PII also includes identities that may be deduced from values like postcode, travel, or multiple locations such as places of purchase. Tiny datasets and rare combinations of values make personal identification easier.

Here some references:
* https://gdpr-info.eu/
* https://www.infoq.com/articles/gdpr-for-software-devs/
* https://www.privacypolicies.com/blog/gdpr-compliance-apps/

omichel, 2020-05-11:
What about other software that are collecting similar anonymous user data such as Firefox or Atom? How do they present this telemetry option to the user?

stefaniapedrazzi, 2020-05-11:
All the documentation I found about GDPR tells that the majority of the software and websites are not compliant with GDPR.

**Firefox:**
* the telemetry data collecting is enabled by default without warning the user at first start
* you can get the complete set of telemetry data gathered on the machine at this URL `about:telemetry`.
* there is an explicit Privacy Notice page: about:preferences#privacy
* data should be deleted 30 days after you disable the telemetry option

**Atom**
At first start it has a "Telemetry Consent" tab with two buttons "Yes" and "No".
If none of the options is chosen, it seems that it sends data anyway.
This is clearly not compliant with GDPR and some users already complained about it. So Atom is the best reference for this.
* https://github.com/atom/atom/issues/20182
* https://www.smalltechnews.com/archives/25668.

Currently Webots is not worse than other software.
We could keep the checkbox as is but maybe add a Privacy Notice (https://gdpr.eu/privacy-notice/).

omichel, 2020-05-11:
Thank you for investigating this. Where do you think would be the best place to add this privacy notice? On the [telemetry page](https://cyberbotics.com/telemetry) itself?

stefaniapedrazzi, 2020-05-11:
I would probably add it in a separate page (that will link to the telemetry page) so that it can be specific to the whole application.

BenjaminDeleze, 2021-11-15:
Fixed in #2223

closed 2021-11-15 by BenjaminDeleze