cyberbotics/webots #1656: use devices in mullti file controller

LeoJhonSong, 2020-05-12:
System: **Ubuntu1804**
Version: **2020b**

Hi, I am trying to split my controller into several files, but it seems that I could not get the `WbDeviceTag` in other files. Here is the outline of my code:

In detection.py:
```python
import time
from controller import GPS

class Detector(object):
    def __init__(self):
        # pass
        self.gps = GPS('gps')
        self.gps.enable(1)

    def run(self):
        while True:
            time.sleep(0.1)
            print(self.gps.getValues())
```
In controller.py:
```python
from controller import Robot

import detection

if __name__ == "__main__":
    robot = Robot()
    detector2 = detect ion.Detector()
    detector2.run()
    timestep = int(robot.getBasicTimeStep())
    while robot.step(timestep) != -1:
        pass
```

Nothing will output when runnning the simulation, until the **Reset Simulation** button is pressed. However, the value of GPS is `[nan, nan, nan]`, not as expected...

I wounder if there is a way to use devices in a multi files controller?

DavidMansolino, 2020-05-12:
The problem is not the multi file.
This line is wrong:
```
 self.gps = GPS('gps')
```
To get a gps handle you should call the ``Robot.getGPS`` function:
https://cyberbotics.com/doc/reference/robot?tab-language=python#wb_robot_get_device

You should probably have a look at this tutorial about how Python controllers works:
https://cyberbotics.com/doc/guide/tutorial-4-more-about-controllers?tab-language=python

LeoJhonSong, 2020-05-12:
> To get a gps handle you should call the `Robot.getGPS` function:
> https://cyberbotics.com/doc/reference/robot?tab-language=python#wb_robot_get_device

Yes I know this official way to do it, but I found `gps = GPS('gps')` also works in normal situations :joy:

> 

I think you are right, the result is same even if the code is like this:
```python
from controller import Robot, GPS
import time

import detection

class Detector(object):
    def __init__(self):
        global robot
        self.gps = robot.getGPS('gps')
        self.gps.enable(1)

    def run(self):
        while True:
            time.sleep(0.1)
            print(self.gps.getValues())

if __name__ == "__main__":
    robot = Robot()
    detector2 = Detector()
    detector2.run()
    timestep = int(robot.getBasicTimeStep())
    while robot.step(timestep) != -1:
        pass
```

DavidMansolino, 2020-05-12:
> Yes I know this official way to do it, but I found gps = GPS('gps') also works in normal situations

Yes it might work, bit this is for internal use only and should be avoided.

You new code looks fine (except for the blocking call to `detector2.run()`), however, when splitting to multi-file the 'global' trick may not work anymore. A more robust mechanism can be the following (untested):

detection.py:
```Python
class Detector(object):
    def __init__(self, robot):
        self.gps = robot.getGPS('gps')
        self.gps.enable(1)

    def run(self):
        print(self.gps.getValues())
```
controller.py:
```Python
from controller import Robot, GPS
import time

from detection import Detector

if __name__ == "__main__":
    robot = Robot()
    detector2 = Detector(robot)
    timestep = int(robot.getBasicTimeStep())
    while robot.step(timestep) != -1:
        detector2.run()
```

LeoJhonSong, 2020-05-12:
> You new code looks fine

Well the problem is it does not work :joy: it still gives me `[nan, nan, nan]`, and the simulation seems to stuck at the beginning. Here is a gif showing what I see.
[image]

DavidMansolino, 2020-05-12:
This is because of the booking call to the detector2.run() function. Have you tried the new version I posted?

LeoJhonSong, 2020-05-12:
Oh yes you are perfectly right! Thank you very much!

closed 2020-05-12 by LeoJhonSong

DavidMansolino, 2020-05-12:
You're welcome!