cyberbotics/webots #1660: wrong length of tab in webots console

LeoJhonSong, 2020-05-13:
**System**: Ubuntu1804
**Version**: 2020b

Hi, I was trying to use '\t' to prettier my output, but I surprisingly found that a horizontal tab in webots console is **11** but not 8 spaces. Is that what supposed to be?
[image]

DavidMansolino, 2020-05-14:
You're right, I tested on macOS it is not equal to 8 spaces either, but rather 7. This probably depends on the font used.

omichel, 2020-05-14:
Indeed. This value is expressed in pixels and the default value is 80 pixels. See https://doc.qt.io/qt-5/qplaintextedit.html#tabStopDistance-prop
I write a small test controller program on Windows:
```C
  char *s = "12345678901234567890";
  char buffer[256];
  printf("              %s%s%s%s%s%s\n", s, s, s, s, s, s);
  for (int i = 0; i < 20; i++) {
    int a[4];
    for (j = 0; j < sizeof(a) / sizeof(int); j++)
      a[j] = rand() % 10;
    sprintf(buffer, "%.2d:\t%%.%ds\t%%.%ds\t%%.%ds\t%%.%ds\n", i + 1, a[0], a[1], a[2], a[3]);
    // puts(buffer);
    printf(buffer, s, s, s, s);
  }
```
And I get this:
[image]
So, I would say the behavior is normal.
Maybe we can add a preferences item for changing this 80 pixels value?

LeoJhonSong, 2020-05-14:
I'm sorry that I have to correct my information:

> [image]

This output above was obtained on **Windows 10**, not Ubuntu1804. So same tab length as what @omichel got. 
Webots console works pretty well on my Ubuntu1804 :smile:

omichel, 2020-05-14:
OK, so I guess we can close this issue.
Feel free to re-open it if you believe there is a problem.

closed 2020-05-14 by omichel

LeoJhonSong, 2020-05-14:
Thank you. BTW are we going to add a preferences item for changing the horizontal tab length?

omichel, 2020-05-14:
If you need it, please go ahead and implement it, I will be happy to review your PR about it.