cyberbotics/webots #1668: webots.min.js: does quit button work as expected?

stefaniapedrazzi, 2020-05-14:
With the changes introduced in #1420, when pressing the Quit button on the web interface the last page is loaded.

Note that if you wrote manually the URL, quitting a robotbenchmark simulation will redirect to the previous page and not the benchmark page.
The previous behavior was much better in case of robotbenchmark.

If the new behavior is really required by some applications, then we should overwrite the `onquit` function for robotbenchmark.

omichel, 2020-05-14:
Yes, I believe it's cleaner to rewrite the *onquit()* method when needed.
The *onquit()* method we provide is now very simple whereas the previous one was more complex and a bit arbitrary. Defaulting to "back" is the most simple thing from a user's perspective. When I quit an application, I want to go back to was I was doing previously, I don't want to land on another page.
So, we should probably keep it by default and overwrite it/if when needed.

stefaniapedrazzi, 2020-05-14:
Ok. I will change it in the robotbenchmark repository.

closed 2020-05-14 by stefaniapedrazzi