cyberbotics/webots #1674: could do nothing after wb_robot_step function returns -1

LeoJhonSong, 2020-05-18:
**System**: Ubuntu18.04
**Version**: 2020b

Hi, I was trying to do some action after `wb_robot_step` function returns -1. I have tried a simple version of [the example given by the document](https://cyberbotics.com/doc/guide/controller-programming?tab-language=python#controller-termination), which should print out a line 'aaaaaaaaaa` right after I press the **reset** button, but it does not.

Here is my controller:
```python
from controller import Robot

TIME_STEP = 32
robot = Robot()
while robot.step(TIME_STEP) != -1:
    print("running")
# Webots triggered termination detected!
print('aaaaaaaaaa')
```

DavidMansolino, 2020-05-19:
You're right, I can reproduce the issue. The command is actually executed, but the output is not displayed by Webots.

omichel, 2020-05-25:
This is the normal behavior, but was not documented as such.
#1698 improves the documentation to clarify this point.

PR #1698 (2020-05-25): Improve controller termination documentation
Fixes #1674. See: https://cyberbotics.com/doc/guide/controller-programming?version=hotfix-improve-controller-termination-doc#controller-termination
closed 2020-05-25 by omichel