cyberbotics/webots #1675: insert omni-wheel robot

Mohammad9966, 2020-05-18:
Hello everyone, I hope you are doing well
I'm trying to insert an Omni-wheel robot, how can I do that <
and can I train my robot in the simulator, I mean can do a machine learning with Matlab on Webots 
and if there any example pls provide me with it 
thx a lot

LeoJhonSong, 2020-05-19:
In [this sample world](https://cyberbotics.com/doc/guide/samples-howto#four_wheels-wbt) a omni wheel robot is provided. [Here](https://cyberbotics.com/doc/guide/sample-webots-applications) is hwo do you reach it. 

And after you go through [this](https://cyberbotics.com/doc/guide/tutorial-4-more-about-controllers?tab-language=matlab), [this](https://cyberbotics.com/doc/guide/controller-programming?tab-language=matlab) and [this](https://cyberbotics.com/doc/guide/matlab?tab-language=matlab) documents you may have an basic idea of how to deal with Matlab.

Hope this helps you.

closed 2020-05-19 by DavidMansolino

Mohammad9966, 2020-05-19:
it's too long to go through the process of making a new Omni-wheel, is there any ready templet or file for it?

DavidMansolino, 2020-05-19:
Unfortunately not, but you can start from the youbot or robotino3 models as example and modify it to suit your needs.

omichel, 2020-05-19:
Or from this [one](https://cyberbotics.com/doc/guide/samples-howto#omni_wheels-wbt).