cyberbotics/webots #1678: Compilation Warning while Creating the Mac Package

DavidMansolino, 2020-05-19:
**Describe the Bug**
The creation of the Mac package raises the following deprecation warnings:
```
# compiling webots_distro.c
webots_distro.c:264:3: warning: 'CC_MD5_Init' is deprecated: first deprecated in
      macOS 10.15 - This function is cryptographically broken and should not be
      used in security contexts. Clients should migrate to SHA256 (or stronger).
      [-Wdeprecated-declarations]
  MD5_Init(&mc);
  ^
webots_distro.c:27:21: note: expanded from macro 'MD5_Init'
#define MD5_Init(a) CC_MD5_Init(a)
                    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/CommonCrypto/CommonDigest.h:133:12: note: 
      'CC_MD5_Init' has been explicitly marked deprecated here
extern int CC_MD5_Init(CC_MD5_CTX *c)
           ^
webots_distro.c:280:5: warning: 'CC_MD5_Update' is deprecated: first deprecated
      in macOS 10.15 - This function is cryptographically broken and should not
      be used in security contexts. Clients should migrate to SHA256
      (or stronger). [-Wdeprecated-declarations]
    MD5_Update(&mc, bufferUnix, lenUnix);
    ^
webots_distro.c:28:29: note: expanded from macro 'MD5_Update'
#define MD5_Update(a, b, c) CC_MD5_Update(a, b, c)
                            ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/CommonCrypto/CommonDigest.h:136:12: note: 
      'CC_MD5_Update' has been explicitly marked deprecated here
extern int CC_MD5_Update(CC_MD5_CTX *c, const void *data, CC_LONG len)
           ^
webots_distro.c:285:3: warning: 'CC_MD5_Final' is deprecated: first deprecated
      in macOS 10.15 - This function is cryptographically broken and should not
      be used in security contexts. Clients should migrate to SHA256
      (or stronger). [-Wdeprecated-declarations]
  MD5_Final(out, &mc);
  ^
webots_distro.c:29:25: note: expanded from macro 'MD5_Final'
#define MD5_Final(a, b) CC_MD5_Final(a, b)
                        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/CommonCrypto/CommonDigest.h:139:12: note: 
      'CC_MD5_Final' has been explicitly marked deprecated here
extern int CC_MD5_Final(unsigned char *md, CC_MD5_CTX *c)
           ^
3 warnings generated.
```

This would be fixed by fixing #571

DavidMansolino, 2020-06-29:
Will be fixed by #571

omichel, 2020-12-02:
Fixed in #2488.

PR #2488 (2020-12-02): Fixed compilation warning of webots_distro.c on mac
Fix #1678.
closed 2020-12-02 by omichel