cyberbotics/webots #1687: cannot use get_id under CameraRecognitionObject when using Python

fiveriddle, 2020-05-22:
**Describe the Bug**
When i want use the get_id of  CameraRecognitionObject when using Python, it occurs the error that the get_id are not in the list, i wonder how could i take the id or the position out?
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: macOS Mojave

omichel, 2020-05-22:
You should write:
```python
    if data.get_id() == 1813:
```

closed 2020-05-22 by omichel

fiveriddle, 2020-05-22:
> You should write:
> 
> ```python
>     if data.get_id() == 1813:
> ```
it  still occur the error
[recog] Traceback (most recent call last):
[recog]   File "recog.py", line 43, in <module>
[recog]     if data.get_id() == 1803:
[recog] AttributeError: 'list' object has no attribute 'get_id'

omichel, 2020-05-22:
Can you please post your complete controller source code?

fiveriddle, 2020-05-22:
> Can you please post your complete controller source code?
from so far it is like
from controller import Robot
from controller import Camera
from controller import CameraRecognitionObject

TIME_STEP = 64

robot = Robot()
camera1=Camera('camera1')
camera1.enable(TIME_STEP)
camera1.recognitionEnable(TIME_STEP)
wheels = []
data=[]
wheelsNames = ['wheel1', 'wheel2', 'wheel3', 'wheel4']
for i in range(4):
    wheels.append(robot.getMotor(wheelsNames[i]))
    wheels[i].setPosition(float('inf'))
    wheels[i].setVelocity(0.0)

while robot.step(timestep) != -1:
    leftSpeed = 0.0
    rightSpeed = 0.0
    cameraData = camera1.getImage()
    number = camera1.getRecognitionNumberOfObjects() 
    data = camera1.getRecognitionObjects()   
    for i in range(number):
        if data.get_id() == 1803:
            leftSpeed = 6.0
            rightSpeed = 6.0
            if((data.get_position())> -3):
                leftSpeed = 0.0
                rightSpeed = 0.0

omichel, 2020-05-22:
Can you try this:
```python
[...]
    data = camera1.getRecognitionObjects()   
    for d in data:
        if d.get_id() == 1803:
[...]
```

fiveriddle, 2020-05-22:
> Can you try this:
> 
> ```python
> [...]
>     data = camera1.getRecognitionObjects()   
>     for d in data:
>         if d.get_id() == 1803:
> [...]
> ```

> Can you try this:
> 
> ```python
> [...]
>     data = camera1.getRecognitionObjects()   
>     for d in data:
>         if d.get_id() == 1803:
> [...]
> ```

My aim is to scan all the objects in the image to check whether there exist the ideal id(which we have test and record before). Once confirmed, I want to get the z, x or y position of the object. ( for the next step). However, when I take use of python, I do not understand how to call database.
1.  Do I need to use the CameraRecognitionObject API? 
Such as directly use
CameraRecognitionObject.get_id() 
or use 
data = camera1.getRecognitionObjects()
and use data.get_id()
2. how to call the position, do i need to built another 3 length array like 
position =[3]
position=data[i].get_position()
and take use of position[2] to use the z-position?
and how could i know the objects' id 
is the class more like 
data[i].get_position[k](in C language)?
how to write it in python?
Thanks a lot.

omichel, 2020-05-22:
Yes, `getRecognitionObjects()` returns an array of `CameraRecognitionObject` which you can access using either the `[]`operator or the `for object in objects` construct which I originally suggested.