cyberbotics/webots #1690: importing omni wheel to webots

Mohammad9966, 2020-05-22:
Hello, everyone.
since I'm trying to build an Omni wheel and it's too long and confusing, can I draw one on solid works and import it to Webots?
thx alot

Mohammad9966, 2020-05-22:
Or it’s easier to draw a full robot and import it to Weboat environment if it possible

DavidMansolino, 2020-05-25:
Yes, this is possible, however, from SolidWoks you will have to export to VRML and then import this VRML file in Webots.
Doing so you will loose the joint/physics/bounding-object/etc. information, so you will have to do a second manual pass in Webots to re-create the hierarchy of your robot (basically import from SolidWorks is good for the visual part, but not for the rest).

closed 2020-05-25 by DavidMansolino