cyberbotics/webots #1702: Render suddenly only showing empty, was working before

yervantk, 2020-05-26:
**Describe the Bug**
The 3D viewport is empty, shows only grey, but is apparently still rendering. I can click on parts of the screen and objects will get selected (see gif)
This was working two days ago, and rendering just fine. It suddenly stopped. The only difference since then is that I have run the Controller SDK in python and Java. 
I restarted the computer too.

**Steps to Reproduce**
1. Open Webots application on OSX
2. Select a world or sample from the webots guided tour
3. Set the OpenGL preferences to any value (I chose Ambient Occlusion: Disable, texture quality: Low, Disable Shadows, Disable Antialiasing)
4. The objects, scripts will load, and everything will run just fine
5. But there will be no visible render

**Expected behavior**
I expect to see the 3D objects render in the view

**Screenshots**
[image]

**System**
 - Operating System: macOS Sierra 10.12.6
 - Graphics Card: Intel Iris Pro 1536 MB

DavidMansolino, 2020-05-26:
Can you try disabling and then re-enabling the 3D view from the Tools menu?

If this still doesn't work, can you retry with a beta version of Webots R2020a-rev2 (we fixed an issue related to this on mac OS for the next version of Webots) available here: https://github.com/cyberbotics/webots/releases

yervantk, 2020-05-26:
I tried disabling/reenabling the 3D View, didn't work. 
I'll try the beta

yervantk, 2020-05-26:
Same problem, behaves the same too

omichel, 2020-05-27:
Can you post a screenshot of what you get when going to the Help menu / OpenGL information?

omichel, 2020-05-27:
Also, I believe that upgrading mac OS from High Sierra to Catalina or Mojave will likely fix the problem.

yervantk, 2020-05-27:
Upgraded to Catalina. You were right, seems to have fixed it for now. You can close this issue, I'll reopen if something goes wrong.
Thanks

closed 2020-05-28 by omichel