cyberbotics/webots #1708: Cannot load textures in WIN10, warning: "Unsupported image format"

shuileng, 2020-05-28:
**Describe the Bug**
I met this issue for the first time on new my computer in WIN10, and my laptop works fine. 

**Steps to Reproduce**
1. Open any project/world with .jpg file
2. See error

**Expected behavior**
Textures should be displayed and no warnings generated.

**Screenshots**
WARNING: Texture image size of 'D:/webots/Webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_base_color.jpg' is not a power of two: rescaling it from -1x-1 to 0x0.
WARNING: Floor (PROTO) > ChequeredParquetry (PROTO) > ImageTexture: Cannot load texture 'D:/webots/Webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_base_color.jpg': Unsupported image format.
WARNING: Texture image size of 'D:/webots/Webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_roughness.jpg' is not a power of two: rescaling it from -1x-1 to 0x0.
WARNING: Floor (PROTO) > ChequeredParquetry (PROTO) > ImageTexture: Cannot load texture 'D:/webots/Webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_roughness.jpg': Unsupported image format.
WARNING: Texture image size of 'D:/webots/Webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_normal.jpg' is not a power of two: rescaling it from -1x-1 to 0x0.
WARNING: Floor (PROTO) > ChequeredParquetry (PROTO) > ImageTexture: Cannot load texture 'D:/webots/Webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_normal.jpg': Unsupported image format.
WARNING: Texture image size of 'D:/webots/Webots/projects/appearances/protos/textures/chequered_parquetry/chequered_parquetry_occlusion.jpg' is not a power of two: rescaling it from -1x-1 to 0x0.

**System**
 - Operating System: WIN 10
 - Graphics Card:  NVIDIA Quadro P620, Intel UHD Graphics 630

omichel, 2020-05-28:
I believe we fixed this problem. Can you download a nightly build of [Webots R2020a revision 2](https://github.com/cyberbotics/webots/releases/download/nightly_26_5_2020/webots-R2020a-rev2_setup.exe) and let me know if that fixes the problem?

shuileng, 2020-05-28:
> I believe we fixed this problem. Can you download a nightly build of [Webots R2020a revision 2](https://github.com/cyberbotics/webots/releases/download/nightly_26_5_2020/webots-R2020a-rev2_setup.exe) and let me know if that fixes the problem?

It works! Thank you so much.

closed 2020-05-28 by shuileng

DrWhoever, 2020-08-25:
the error is back in R2020b deb version. no textures

> **Describe the Bug**
> I met this issue for the first time on new my computer in WIN10, and my laptop works fine.
>
> **Steps to Reproduce**
>
>     1. Open any project/world with .jpg file
>
>     2. See [...]

the error is back in r2020b debian version. no texture file is working at all

DavidMansolino, 2020-08-25:
Can you please make sure that you don't have any conflicting ``libjpeg`` library in your PATH?

DrWhoever, 2020-08-25:
> Can you please make sure that you don't have any conflicting `libjpeg` library in your PATH?

i haven't found any. Honestly, i'm still new to webots, any help would be appreciated

DrWhoever, 2020-08-25:
I finally figured it out. i uninstalled webots, installed libjpeg8 and then reinstalled it again, the textures are now rendering. Thanks!!

DavidMansolino, 2020-08-26:
Perfect, thank you for the feedback!

timonegk, 2021-03-09:
A am on Debian Testing where libjpeg8 is not available and have Webots R2021a installed from the deb. I had to copy the libjpeg.so.8.1.2 and libjpeg.so.8 from [this ubuntu package](https://packages.ubuntu.com/bionic/amd64/libjpeg-turbo8/download) to `/usr/local/webots/lib/webots/` to get rid of this error.