cyberbotics/webots #1712: Broken Controller Arguments

DavidMansolino, 2020-05-29:
**Describe the Bug**
The controller arguments are broken (at least in python).

**Steps to Reproduce**
1. Open and run the pedestrian world.

**Screenshots**
The starting command line looks wrong:
```
INFO: pedestrian: Starting controller: python!-u+pedestrian.py+--trajectory="0+3,+0+-3"+--speed=2.5
```

The controllers fails:
```
[pedestrian] Usage: pedestrian.py [options]
[pedestrian] 
[pedestrian] pedestrian.py: error: no such option: -3
WARNING: 'pedestrian' controller exited with status: 2.
[pedestrian] Usage: pedestrian.py [options]
```

This has probably been introduced in #1709.

DavidMansolino, 2020-05-29:
In this case, the following argument `--trajectory="0 3, 0 -3" --speed=2.5` gives these arguments in the controller (printed with `print(sys.argv)`):
```
[pedestrian] ['pedestrian.py', '--trajectory="0', '3,', '0', '-3"', '--speed=2.5']
```

DavidMansolino, 2020-05-29:
I confirm it was working fine before #1709.

omichel, 2020-05-29:
I believe we have a design problem here. `Robot.controllerArgs` should be a MFString instead of a SFString. I will consider fixing this on the develop branch later. I will now fix the bug with a somehow dirty patch, so that it behaves as before.

PR #1713 (2020-06-02): Fixed controller args with double quotes
Fixes #1712.

Note: This is a quick & dirty fix until we introduce `Robot.controllerArgs` as a `MFString` instead of a `SFString` that will provide a much better interface.
closed 2020-06-02 by omichel