cyberbotics/webots #1734: How to insert a page into the user guide

PRUBHTEJ, 2020-06-07:
**Documentation page(s) concerned**
https://cyberbotics.com/doc/guide/index

**What is the problem with the current documentation?**
*I want to insert a pre-requisites page in the user guide*
So, what is the procedure to insert a page in the user guide at a specific position ( Say between "Building Webots" & "Getting started with webots") ?

DavidMansolino, 2020-06-08:
First, you have to add a new page as a `.md` file here: https://github.com/cyberbotics/webots/tree/master/docs/guide

And then you have to add it to the menu:
  - https://github.com/cyberbotics/webots/blob/master/docs/guide/menu.md

Finally, when you will open a Pull-request on this repo, our CI will run some tests and let you know if there are some issues to fix.

PRUBHTEJ, 2020-06-09:
I have tried adding a page to the user guide, as explained above. I am getting failed checks.
Kindly help.

lukicdarkoo, 2020-06-09:
You can check for details in the error log by clicking `Details` next to the red X and then `The build`.

For example, this is the error log for https://github.com/cyberbotics/webots/pull/1749:
https://travis-ci.com/github/cyberbotics/webots/builds/170536941

If you look carefully you will see this part:
```
FAIL: test_references_are_valid (test_references.TestReferences)
Test that the MD files refer valid URLs.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/travis/build/cyberbotics/webots/docs/tests/test_references.py", line 146, in test_references_are_valid
    self.assertTrue(
AssertionError: False is not true : /home/travis/build/cyberbotics/webots/docs/guide/menu.md: "/home/travis/build/cyberbotics/webots/docs/guide/Pre-requisites.md" not found
```
which means that `/docs/guide/Pre-requisites.md` doesn't exist even though you added a link to the page.

PRUBHTEJ, 2020-06-09:
Thanks for the reply. I had made a PR first for adding the file" pre-requisites.md" to /docs/guide/Pre-requisites.md. This is a https://github.com/cyberbotics/webots/pull/1748 is reference to that which also failed the tests. Then I had added "Pre-requisites.md" to the menu.

omichel, 2020-06-09:
You should both in the same PR.

PRUBHTEJ, 2020-06-09:
I have tried to follow the steps as instructed by @DavidMansolino, but still I'm getting build errors in https://github.com/cyberbotics/webots/pull/1749 and https://github.com/cyberbotics/webots/pull/1748. Kindly guide me how to correct these errors?

PRUBHTEJ, 2020-06-10:
What I have tried to do is, to add a blank page initially to the user guide in #1755 and I'm getting this error:
FAIL: test_all_book_files_are_referred_by_the_menu_file (test_menus.TestMenus)
All the book files are referred to by the menu files.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\projects\webots\docs\tests\test_menus.py", line 84, in test_all_book_files_are_referred_by_the_menu_file
    self.assertIn(

Even adding a blank file is giving me an error. 
How can I correct it?