cyberbotics/webots #1767: GUI is turning gray when loading world with added wrl model

Majanao, 2020-06-11:
**Describe the Bug**
I'm able to import a model based on a wrl files that I created with CATIA or Meshmixer. I'm also able to scale transform or rotate that model. But if I'm trying to reload this world, the GUI shows me the world for a blink of an eye and than it turns gray as if there is no world loaded. Probably the world model crashes and closes.

**Steps to Reproduce**
1) Open new world
2) Insert model with wrl file
3) Save world
4) Load world -> crash

**Expected behavior**
It should load the world

**Screenshots**
The wrl files I used where created with CATIA or Meshmixer:
[wrl files.zip](https://github.com/cyberbotics/webots/files/4764205/wrl.files.zip)

**System**
 - Operating System: Windows 10
- Graphics: Geforce MX150

**Additional context**
Add any other context about the problem here.

DavidMansolino, 2020-06-11:
For me it seems to work fine.
May I ask you to try the following:
  - try with a beta of Webots R2020a-revision2: https://github.com/cyberbotics/webots/releases
  - try with the safe mode enabled: https://cyberbotics.com/doc/guide/starting-webots#on-windows

omichel, 2020-06-29:
I am closing this issue as we didn't received any answer from @Majanao. Feel free to re-open it if needed.

closed 2020-06-29 by omichel