cyberbotics/webots #1769: QtWheelEvent::delta() is deprecated

stefaniapedrazzi, 2020-06-11:
On Ubuntu 18.04 I'm getting these warnings when compiling Webots:

```
gui/WbView3D.cpp: In member function ‘virtual void WbView3D::wheelEvent(QWheelEvent*)’:
gui/WbView3D.cpp:2212:26: warning: ‘Qt::Orientation QWheelEvent::orientation() const’ is deprecated: Use angleDelta() [-Wdeprecated-declarations]
   if (event->orientation() != Qt::Vertical)
                          ^
In file included from ../../include/qt/QtGui/QtGui/QKeyEvent:1:0,
                 from gui/WbView3D.cpp:76:
../../include/qt/QtGui/QtGui/qevent.h:222:28: note: declared here
     inline Qt::Orientation orientation() const { return qt4O; }
                            ^~~~~~~~~~~
gui/WbView3D.cpp:2221:34: warning: ‘int QWheelEvent::delta() const’ is deprecated: Use angleDelta() [-Wdeprecated-declarations]
       mWheel->apply(event->delta());
                                  ^
In file included from ../../include/qt/QtGui/QtGui/QKeyEvent:1:0,
                 from gui/WbView3D.cpp:76:
../../include/qt/QtGui/QtGui/qevent.h:219:16: note: declared here
     inline int delta() const  { return qt4D; }
                ^~~~~
gui/WbView3D.cpp:2231:32: warning: ‘int QWheelEvent::delta() const’ is deprecated: Use angleDelta() [-Wdeprecated-declarations]
     mWheel->apply(event->delta());
                                ^
In file included from ../../include/qt/QtGui/QtGui/QKeyEvent:1:0,
                 from gui/WbView3D.cpp:76:
../../include/qt/QtGui/QtGui/qevent.h:219:16: note: declared here
     inline int delta() const  { return qt4D; }
                ^~~~~
gui/WbView3D.cpp:2236:24: warning: ‘int QWheelEvent::delta() const’ is deprecated: Use angleDelta() [-Wdeprecated-declarations]
       if (event->delta() > 0)
                        ^
In file included from ../../include/qt/QtGui/QtGui/QKeyEvent:1:0,
                 from gui/WbView3D.cpp:76:
../../include/qt/QtGui/QtGui/qevent.h:219:16: note: declared here
     inline int delta() const  { return qt4D; }
                ^~~~~
gui/WbView3D.cpp:2255:54: warning: ‘int QWheelEvent::delta() const’ is deprecated: Use angleDelta() [-Wdeprecated-declarations]
     const double scaleFactor = -0.02 * (event->delta() < 0.0 ? -1 : 1) * distanceToPickPosition;
                                                      ^
In file included from ../../include/qt/QtGui/QtGui/QKeyEvent:1:0,
                 from gui/WbView3D.cpp:76:
../../include/qt/QtGui/QtGui/qevent.h:219:16: note: declared here
     inline int delta() const  { return qt4D; }

```

DavidMansolino, 2020-06-11:
That's strange that we didn't get any warnings in the nightly build on Ubuntu 18.04: https://travis-ci.com/github/cyberbotics/webots/jobs/347319923#L3138

stefaniapedrazzi, 2020-06-11:
Yes it's strange. I also saw them yesterday for the first time when working on the `WbView3D` class.
But I checked on the Qt documentation and the warnings seem correct:
https://doc.qt.io/qt-5/qwheelevent.html
https://doc.qt.io/qt-5/qwheelevent-obsolete.html

stefaniapedrazzi, 2020-06-11:
In this travis test the warnings are reported:
https://travis-ci.com/github/cyberbotics/webots/jobs/347528912

DavidMansolino, 2020-06-11:
OK, maybe it depends on the make target (distrib/debug/release), but not sure this can explain it.

stefaniapedrazzi, 2020-06-11:
Not sure if this is correct, but these warnings are not making the travis test fail.

omichel, 2020-06-11:
The problem was introduced in #1754.

stefaniapedrazzi, 2020-06-11:
The warnings fix is reintroduced in #1770

closed 2020-06-11 by stefaniapedrazzi