cyberbotics/webots #1774: XCB plugin issue at startup

nsevalia, 2020-06-11:
I have just installed the webot latest version on Ubuntu 18.04 LTS platform. Am following the getting started documentation. On starting the webot server am getting the following fatal error:

root@webot-instance:~# webots
Info: Could not load the Qt platform plugin "xcb" in "/usr/local/webots/lib/webots/qt/plugins" even though it was found.
Fatal: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

/usr/local/bin/webots: line 86: 11986 Aborted                 (core dumped) "$webots_home/bin/webots-bin" "$@"_

Is reinstalling the application the only way forward or is there any other solution?

DavidMansolino, 2020-06-12:
Re-installing the application will probably not help.
Can you please give us the result of the following commands:
```
export LD_LIBRARY_PATH=/usr/local/webots/lib/webots
ldd /usr/local/webots/bin/webots-bin
```
Thank you.

nsevalia, 2020-06-12:
> <author@noreply.github.com>
> (#1774)
>
> "External email. Open with Caution"

Version = R2020a-rev1 

Here is the output from ldd command.

linux-vdso.so.1 (0x00007fff373fd000)
libQt5Core.so.5 => /usr/local/webots/lib/webots/libQt5Core.so.5 
(0x00007fd112a68000)
libQt5Network.so.5 => /usr/local/webots/lib/webots/libQt5Network.so.5 
(0x00007fd1126b8000)
libQt5Gui.so.5 => /usr/local/webots/lib/webots/libQt5Gui.so.5 
(0x00007fd111e6c000)
libQt5OpenGL.so.5 => /usr/local/webots/lib/webots/libQt5OpenGL.so.5 
(0x00007fd111c15000)
libQt5WebSockets.so.5 => 
/usr/local/webots/lib/webots/libQt5WebSockets.so.5 (0x00007fd1119e8000)
libQt5Widgets.so.5 => /usr/local/webots/lib/webots/libQt5Widgets.so.5 
(0x00007fd111194000)
libQt5PrintSupport.so.5 => 
/usr/local/webots/lib/webots/libQt5PrintSupport.so.5 (0x00007fd110f22000)
libQt5WebEngineWidgets.so.5 => 
/usr/local/webots/lib/webots/libQt5WebEngineWidgets.so.5 
(0x00007fd110cde000)
libQt5WebChannel.so.5 => 
/usr/local/webots/lib/webots/libQt5WebChannel.so.5 (0x00007fd110abe000)
libQt5Xml.so.5 => /usr/local/webots/lib/webots/libQt5Xml.so.5 
(0x00007fd110884000)
libopenal.so.1 => /usr/local/webots/lib/webots/libopenal.so.1 
(0x00007fd110618000)
libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007fd11038c000)
libGLU.so.1 => /usr/lib/x86_64-linux-gnu/libGLU.so.1 (0x00007fd11011d000)
libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fd10ff19000)
libOIS-1.4.0.so => /usr/local/webots/lib/webots/libOIS-1.4.0.so 
(0x00007fd10fcf8000)
libpico.so => /usr/local/webots/lib/webots/libpico.so (0x00007fd10fa87000)
libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 
(0x00007fd10f7d3000)
libassimp.so.3 => /usr/local/webots/lib/webots/libassimp.so.3 
(0x00007fd10ef44000)
librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fd10ed3c000)
libode.so => /usr/local/webots/lib/webots/libode.so (0x00007fd10e97e000)
libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 
(0x00007fd10e5f5000)
libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd10e257000)
libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd10e03f000)
libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd10dc4e000)
libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 
(0x00007fd10da2f000)
libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fd10d812000)
libicui18n.so.56 => /usr/local/webots/lib/webots/libicui18n.so.56 
(0x00007fd10d379000)
libicuuc.so.56 => /usr/local/webots/lib/webots/libicuuc.so.56 
(0x00007fd10cfc1000)
libicudata.so.56 => /usr/local/webots/lib/webots/libicudata.so.56 
(0x00007fd10b5de000)
libgthread-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgthread-2.0.so.0 
(0x00007fd10b3dc000)
libglib-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0 
(0x00007fd10b0c5000)
/lib64/ld-linux-x86-64.so.2 (0x00007fd113217000)
libQt5WebEngineCore.so.5 => 
/usr/local/webots/lib/webots/libQt5WebEngineCore.so.5 (0x00007fd103442000)
libQt5Quick.so.5 => /usr/local/webots/lib/webots/libQt5Quick.so.5 
(0x00007fd102d9d000)
libQt5Qml.so.5 => /usr/local/webots/lib/webots/libQt5Qml.so.5 
(0x00007fd1026fd000)
libQt5Positioning.so.5 => 
/usr/local/webots/lib/webots/libQt5Positioning.so.5 (0x00007fd102461000)
libQt5QuickWidgets.so.5 => 
/usr/local/webots/lib/webots/libQt5QuickWidgets.so.5 (0x00007fd10224e000)
libGLX.so.0 => /usr/lib/x86_64-linux-gnu/libGLX.so.0 (0x00007fd10201d000)
[... 45 lines cut ...]
(0x00007fd0fe139000)
libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 
(0x00007fd0fdeb5000)
libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007fd0fdca0000)
liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007fd0fda7a000)
liblz4.so.1 => /usr/lib/x86_64-linux-gnu/liblz4.so.1 (0x00007fd0fd85e000)
libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 
(0x00007fd0fd542000)
libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 
(0x00007fd0fd32d000)

Best Regards
Naishadh Sevalia

Notice: The information contained in this e-mail
message and/or attachments to it may contain 
confidential or privileged information. If you are 
not the intended recipient, any dissemination, use, 
review, distribution, printing or copying of the 
information contained in this e-mail message 
and/or attachments to it are strictly prohibited. If 
you have received this communication in error, 
please notify us by reply e-mail or telephone and 
immediately and permanently delete the message 
and any attachments. Thank you

DavidMansolino, 2020-06-15:
Ok, everything looks good in this log.
Can you please give me the output of these other commands?
```
export LD_LIBRARY_PATH=/usr/local/webots/lib/webots
ldd /usr/local/webots/qt/plugins/platforms/libqxcb.so
```

Thank you.

nsevalia, 2020-06-16:
Here is the output... the plugin file seems missing!!

root@webot-instance:/home/pnsevalia# export LD_LIBRARY_PATH=/usr/local/webots/lib/webots
root@webot-instance:/home/pnsevalia# ldd /usr/local/webots/qt/plugins/platforms/libqxcb.so
ldd: /usr/local/webots/qt/plugins/platforms/libqxcb.so: No such file or directory
root@webot-instance:/home/pnsevalia#

DavidMansolino, 2020-06-16:
Sorry I missed a `lib` folder, the correct command is:
```
export LD_LIBRARY_PATH=/usr/local/webots/lib/webots
ldd /usr/local/webots/lib/webots/qt/plugins/platforms/libqxcb.so
``

nsevalia, 2020-06-16:
Here is the output for
export LD_LIBRARY_PATH=/usr/local/webots/lib/webots
ldd /usr/local/webots/lib/webots/qt/plugins/platforms/libqxcb.so

	linux-vdso.so.1 (0x00007ffd28957000)
	libQt5XcbQpa.so.5 => /usr/local/webots/lib/webots/libQt5XcbQpa.so.5 (0x00007f65960c7000)
	libfontconfig.so.1 => /usr/lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007f6595e82000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f6595bce000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f65959b1000)
	libQt5Gui.so.5 => /usr/local/webots/lib/webots/libQt5Gui.so.5 (0x00007f6595165000)
	libQt5DBus.so.5 => /usr/local/webots/lib/webots/libQt5DBus.so.5 (0x00007f6594ed9000)
	libQt5Core.so.5 => /usr/local/webots/lib/webots/libQt5Core.so.5 (0x00007f659472a000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007f659449e000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f659427f000)
	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007f659407d000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f6593e55000)
	libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007f6593c4b000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f6593a39000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f6593701000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f6593363000)
	libxkbcommon-x11.so.0 => not found
	libxkbcommon.so.0 => /usr/lib/x86_64-linux-gnu/libxkbcommon.so.0 (0x00007f6593124000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f6592f20000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f6592b97000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f659297f000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f659258e000)
	libxkbcommon-x11.so.0 => not found
	libgthread-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgthread-2.0.so.0 (0x00007f659238c000)
	libglib-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f6592075000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f6591e43000)
	libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f6591c11000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007f65919c4000)
	libicui18n.so.56 => /usr/local/webots/lib/webots/libicui18n.so.56 (0x00007f659152b000)
	libicuuc.so.56 => /usr/local/webots/lib/webots/libicuuc.so.56 (0x00007f6591173000)
	libicudata.so.56 => /usr/local/webots/lib/webots/libicudata.so.56 (0x00007f658f790000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f659666c000)
	libGLX.so.0 => /usr/lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f658f55f000)
	libGLdispatch.so.0 => /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f658f2a9000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f658f0a5000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f658ee9f000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f658ec2d000)
	libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007f658e9a9000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f658e794000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f658e58c000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f658e366000)
	liblz4.so.1 => /usr/lib/x86_64-linux-gnu/liblz4.so.1 (0x00007f658e14a000)
	libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007f658de2e000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f658dc19000)

Thanks  David for helping on  this.

DavidMansolino, 2020-06-16:
Ok, it seems you are missing one of the dependencies, can you try installing the `libxkbcommon-x11-dev` package?
```sudo apt install libxkbcommon-x11-dev```

nsevalia, 2020-06-16:
I installed the package with your suggested command.
It successfully got installed.
Then I tried to start "webots" and got the same fatal error again.

root@webot-instance:/home/pnsevalia# webots
Warning: could not connect to display 
Info: Could not load the Qt platform plugin "xcb" in "/usr/local/webots/lib/webots/qt/plugins" even though it was found.
Fatal: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

/usr/local/bin/webots: line 86:  2900 Aborted                 (core dumped) "$webots_home/bin/webots-bin" "$@"

DavidMansolino, 2020-06-16:
Is the following commands still giving the same output?
```export LD_LIBRARY_PATH=/usr/local/webots/lib/webots
ldd /usr/local/webots/lib/webots/qt/plugins/platforms/libqxcb.so``

nsevalia, 2020-06-16:
The output is different... there are double libs listed now (earlier there were 44 now its 88).
Here  is  the output..

	linux-vdso.so.1 (0x00007ffd28957000)
	libQt5XcbQpa.so.5 => /usr/local/webots/lib/webots/libQt5XcbQpa.so.5 (0x00007f65960c7000)
	libfontconfig.so.1 => /usr/lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007f6595e82000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f6595bce000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f65959b1000)
	libQt5Gui.so.5 => /usr/local/webots/lib/webots/libQt5Gui.so.5 (0x00007f6595165000)
	libQt5DBus.so.5 => /usr/local/webots/lib/webots/libQt5DBus.so.5 (0x00007f6594ed9000)
	libQt5Core.so.5 => /usr/local/webots/lib/webots/libQt5Core.so.5 (0x00007f659472a000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007f659449e000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f659427f000)
	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007f659407d000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f6593e55000)
	libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007f6593c4b000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f6593a39000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f6593701000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f6593363000)
	libxkbcommon-x11.so.0 => not found
	libxkbcommon.so.0 => /usr/lib/x86_64-linux-gnu/libxkbcommon.so.0 (0x00007f6593124000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f6592f20000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f6592b97000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f659297f000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f659258e000)
	libxkbcommon-x11.so.0 => not found
	libgthread-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgthread-2.0.so.0 (0x00007f659238c000)
	libglib-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f6592075000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f6591e43000)
	libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f6591c11000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007f65919c4000)
	libicui18n.so.56 => /usr/local/webots/lib/webots/libicui18n.so.56 (0x00007f659152b000)
	libicuuc.so.56 => /usr/local/webots/lib/webots/libicuuc.so.56 (0x00007f6591173000)
	libicudata.so.56 => /usr/local/webots/lib/webots/libicudata.so.56 (0x00007f658f790000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f659666c000)
	libGLX.so.0 => /usr/lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f658f55f000)
	libGLdispatch.so.0 => /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f658f2a9000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f658f0a5000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f658ee9f000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f658ec2d000)
	libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007f658e9a9000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f658e794000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f658e58c000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f658e366000)
	liblz4.so.1 => /usr/lib/x86_64-linux-gnu/liblz4.so.1 (0x00007f658e14a000)
	libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007f658de2e000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f658dc19000)
	linux-vdso.so.1 (0x00007ffca6fd1000)
	libQt5XcbQpa.so.5 => /usr/local/webots/lib/webots/libQt5XcbQpa.so.5 (0x00007fd5394ee000)
	libfontconfig.so.1 => /usr/lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007fd5392a9000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007fd538ff5000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fd538dd8000)
	libQt5Gui.so.5 => /usr/local/webots/lib/webots/libQt5Gui.so.5 (0x00007fd53858c000)
	libQt5DBus.so.5 => /usr/local/webots/lib/webots/libQt5DBus.so.5 (0x00007fd538300000)
	libQt5Core.so.5 => /usr/local/webots/lib/webots/libQt5Core.so.5 (0x00007fd537b51000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007fd5378c5000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd5376a6000)
	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007fd5374a4000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fd53727c000)
	libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007fd537072000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007fd536e60000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007fd536b28000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd53678a000)
	libxkbcommon-x11.so.0 => /usr/lib/x86_64-linux-gnu/libxkbcommon-x11.so.0 (0x00007fd536582000)
	libxkbcommon.so.0 => /usr/lib/x86_64-linux-gnu/libxkbcommon.so.0 (0x00007fd536343000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fd53613f000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd535db6000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd535b9e000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd5357ad000)
	libgthread-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgthread-2.0.so.0 (0x00007fd5355ab000)
	libglib-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fd535294000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007fd535062000)
	libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007fd534e30000)
[... cut ...]
	libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007fd5319ac000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007fd531797000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fd53158f000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007fd531369000)
	liblz4.so.1 => /usr/lib/x86_64-linux-gnu/liblz4.so.1 (0x00007fd53114d000)
	libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007fd530e31000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007fd530c1c000)

DavidMansolino, 2020-06-16:
It's strange it say once that `libxkbcommon-x11` is not found and once that it is found.
To compare here is the output I have:
```
david@david-UX310UQK:~/webots$ ldd lib/webots/qt/plugins/platforms/libqxcb.so 
	linux-vdso.so.1 (0x00007ffc844a3000)
	libQt5XcbQpa.so.5 => /home/david/webots/lib/webots/libQt5XcbQpa.so.5 (0x00007f57f18d4000)
	libfontconfig.so.1 => /lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007f57f1851000)
	libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f57f1792000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f57f1776000)
	libQt5Gui.so.5 => /home/david/webots/lib/webots/libQt5Gui.so.5 (0x00007f57f0e42000)
	libQt5DBus.so.5 => /home/david/webots/lib/webots/libQt5DBus.so.5 (0x00007f57f0bb6000)
	libQt5Core.so.5 => /home/david/webots/lib/webots/libQt5Core.so.5 (0x00007f57f03c1000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007f57f0339000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f57f0316000)
	libX11-xcb.so.1 => /lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007f57f0311000)
	libxcb-icccm.so.4 => /lib/x86_64-linux-gnu/libxcb-icccm.so.4 (0x00007f57f030a000)
	libxcb-image.so.0 => /lib/x86_64-linux-gnu/libxcb-image.so.0 (0x00007f57f0105000)
	libxcb-shm.so.0 => /lib/x86_64-linux-gnu/libxcb-shm.so.0 (0x00007f57f00fe000)
	libxcb-keysyms.so.1 => /lib/x86_64-linux-gnu/libxcb-keysyms.so.1 (0x00007f57f00f9000)
	libxcb-randr.so.0 => /lib/x86_64-linux-gnu/libxcb-randr.so.0 (0x00007f57f00e6000)
	libxcb-render-util.so.0 => /lib/x86_64-linux-gnu/libxcb-render-util.so.0 (0x00007f57f00df000)
	libxcb-render.so.0 => /lib/x86_64-linux-gnu/libxcb-render.so.0 (0x00007f57f00d0000)
	libxcb-shape.so.0 => /lib/x86_64-linux-gnu/libxcb-shape.so.0 (0x00007f57f00cb000)
	libxcb-sync.so.1 => /lib/x86_64-linux-gnu/libxcb-sync.so.1 (0x00007f57f00bf000)
	libxcb-xfixes.so.0 => /lib/x86_64-linux-gnu/libxcb-xfixes.so.0 (0x00007f57f00b5000)
	libxcb-xinerama.so.0 => /lib/x86_64-linux-gnu/libxcb-xinerama.so.0 (0x00007f57f00b0000)
	libxcb-xkb.so.1 => /lib/x86_64-linux-gnu/libxcb-xkb.so.1 (0x00007f57f0092000)
	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f57f0068000)
	libXext.so.6 => /lib/x86_64-linux-gnu/libXext.so.6 (0x00007f57f0053000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007f57eff14000)
	libxkbcommon-x11.so.0 => /lib/x86_64-linux-gnu/libxkbcommon-x11.so.0 (0x00007f57eff09000)
	libxkbcommon.so.0 => /lib/x86_64-linux-gnu/libxkbcommon.so.0 (0x00007f57efec7000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f57efec1000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f57efce0000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f57efb91000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f57efb74000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f57ef982000)
	libgthread-2.0.so.0 => /lib/x86_64-linux-gnu/libgthread-2.0.so.0 (0x00007f57ef97d000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f57ef854000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f57ef826000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007f57ef81b000)
	libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f57ef7e3000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007f57ef792000)
	libicui18n.so.56 => /home/david/webots/lib/webots/libicui18n.so.56 (0x00007f57ef2f9000)
	libicuuc.so.56 => /home/david/webots/lib/webots/libicuuc.so.56 (0x00007f57eef41000)
	libicudata.so.56 => /home/david/webots/lib/webots/libicudata.so.56 (0x00007f57ed55c000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f57f1e31000)
	libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f57ed4a4000)
	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f57ed470000)
	libxcb-util.so.1 => /lib/x86_64-linux-gnu/libxcb-util.so.1 (0x00007f57ed26a000)
	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007f57ed264000)
	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f57ed25a000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f57ed1e7000)
	libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007f57ed13a000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f57ed120000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f57ed113000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f57ed0ec000)
	liblz4.so.1 => /lib/x86_64-linux-gnu/liblz4.so.1 (0x00007f57ed0cb000)
	libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007f57ecfad000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f57ecf8a000)
``

DavidMansolino, 2020-06-16:
Are you using a desktop or server image of Ubuntu 18.04 LTS?

nsevalia, 2020-06-16:
Ubuntu 18.04 LTS server image

Best Regards
Naishadh Sevalia
TCS - Garima Park,
Gandhinagar, Gujarat, India
Stay Home, Stay safe, Stay Healthy

Date:   16-06-2020 12:43
(#1774)

DavidMansolino, 2020-06-16:
Ok, then this is the cause of the issue.
You need to install some other additional packages: https://cyberbotics.com/doc/guide/installation-procedure#server-edition

nsevalia, 2020-06-19:
Hi David,

I tried to install the server edition packages you had referred to. However  they seem to be already installed. Still the fatal error pops up on the webots  process start  up.

root@webot-instance:/home/pnsevalia# apt install xserver-xorg-core
Reading package lists... Done
Building dependency tree       
Reading state information... Done
xserver-xorg-core is already the newest version (2:1.19.6-1ubuntu4.4).
xserver-xorg-core set to manually installed.
The following package was automatically installed and is no longer required:
  grub-pc-bin
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.
root@webot-instance:/home/pnsevalia# 
root@webot-instance:/home/pnsevalia# 
root@webot-instance:/home/pnsevalia# apt install libpulse0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libpulse0 is already the newest version (1:11.1-1ubuntu7.8).
libpulse0 set to manually installed.
The following package was automatically installed and is no longer required:
  grub-pc-bin
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.
root@webot-instance:/home/pnsevalia# 
root@webot-instance:/home/pnsevalia# 
root@webot-instance:/home/pnsevalia# webots
Warning: could not connect to display 
Info: Could not load the Qt platform plugin "xcb" in "/usr/local/webots/lib/webots/qt/plugins" even though it was found.
Fatal: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

/usr/local/bin/webots: line 86:  3208 Aborted                 (core dumped) "$webots_home/bin/webots-bin" "$@"
root@webot-instance:/home/pnsevalia#

DavidMansolino, 2020-06-19:
Ok, just to make sure you have all the required packages can you try installing all these packages:
```
xserver-xorg-core libpulse0 libglu1-mesa g++ cmake libusb-dev swig python2.7-dev libglu1-mesa-dev libglib2.0-dev libfreeimage-dev libfreetype6-dev libxml2-dev libzzip-0-13 libssl1.0-dev libboost-dev libavcodec-extra libgd-dev libssh-gcrypt-dev libzip-dev python-pip libreadline-dev libassimp-dev ffmpeg npm python3.6-dev python3.7-dev openjdk-11-jdk curl fakeroot pbzip2 libxkbcommon-x11-dev libxcb-keysyms1 libxcb-image0 libxcb-icccm4 libxcb-randr0 libxcb-render-util0 libxcb-xinerama0
```
And it still doesn't work, please try running Webots with a virtual screen:
```
sudo apt install xvfb
xvfb-run --auto-servernum webots --stdout--stderr --minimize --batch
```

DavidMansolino, 2020-06-29:
Please re-open this PR if you have more information.

closed 2020-06-29 by DavidMansolino