cyberbotics/webots #1778: does not work robotis-op2 camera after simulation reset

xxchenchen, 2020-06-12:
**Describe the Bug**
A clear and concise description of what the bug is.

**Steps to Reproduce**
1. ' generate a controller and a supervisor for op2. '
2. 'use your new simulation reset command in supervisor'
3. run for your controller for one step then see reset and the first position of robot.
4. Check the camera image. It is one before reset. Please see it. 
5. I can see new images in screen. However I cant receive no more change in the camera array. 
by using camera.getImageArray()).

**Expected behavior**
After reset, it is expected to change the images taken by using camera.getImageArray())

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: [e.g., Windows 10, Linux Ubuntu 18.04, macOS Mojave]
 - Graphics Card: [e.g., NVIDIA GeForce RTX 2080 11 GB, AMD Radeon RX 580 8GB, etc.]

**Additional context**
Add any other context about the problem here.

DavidMansolino, 2020-06-15:
I have tried reproducing the problem with this controller:
```Python
from controller import Robot

robot = Robot()

timestep = int(robot.getBasicTimeStep())

camera = robot.getCamera('Camera')
camera.enable(timestep)

while robot.step(timestep) != -1:
    print(camera.getImageArray()[100][100])
```

Unfortunately, it seems to work fine (the printed value changes when moving the robot even after a reset).
Can you give us a bit more information?
  - Which version of Webots are you using?
  - Which language for the controller?
  - How do you reset the simulation, from the GUI or from a Supervisor?
  - Can you share your controller/simulation?

xxchenchen, 2020-06-15:
Thanks for answer.

I am sure that it does not work. I still check whether or not there is a my error.

It is nightly build 10/06/2020.

I have python supervisor and python controller.  I have a robot node. I am using supervisor = self.simulationReset()

xxchenchen, 2020-06-15:
I am sure. There is a new camera image on screen but there is not new camera array after reset.

self.getCamera('Camera');
self.camera.enable(self.timeStep);
self.image = np.array(self.camera.getImageArray());

xxchenchen, 2020-06-17:
self.simulationReset()

DavidMansolino, 2020-06-17:
Ok, I can indeed reproduce the issue on the develop branch when the reset is done from a supervisor and the controller is not restarted:
```
from controller import Supervisor

supervisor = Supervisor()

timestep = int(supervisor.getBasicTimeStep())

camera = supervisor.getCamera('Camera')
camera.enable(timestep)

while supervisor.step(timestep) != -1:
    print(camera.getImageArray()[100][100])
    if supervisor.getTime() > 8.0:
        supervisor.simulationReset()
        # supervisor.getSelf().restartController()  # this fixes the issue
```

xxchenchen, 2020-06-17:
Many Thanks. 

The part, "self.getCamera('Camera');
self.camera.enable(self.timeStep);
self.image = np.array(self.camera.getImageArray());" is in my CONTROLLER file .

and 

the part,"          if self.receiver.getQueueLength()>0:
           
            message = self.receiver.getData().decode('utf-8')
            self.receiver.nextPacket()
           
            if message == 'reset':
                .self.simulationReset()

" is  in SUPERVISOR file.

Moreover,
I am not using "# supervisor.getSelf().restartController() "

PR #1802 (2020-06-19): Fix Camera Reset
**Description**
Fix #1778.
closed 2020-06-19 by DavidMansolino