cyberbotics/webots #1787: Both Java examples return null-pointer exception when running at line 15

MichaelFrancis123, 2020-06-15:
**Documentation page(s) concerned**
https://cyberbotics.com/doc/guide/supervisor-programming?tab-language=java

The Java examples do build, but when running the simulation there is a null-pointer exception on line 15. This line is:
    final Field trans_field = robot_node.getField("translation");

I would like an example code which I can run on a robot object.

MichaelFrancis123, 2020-06-15:
I have noticed that I have this issue with other languages too, which makes me seem to think it hasn't to do with the language, but with my robot settings. Of course I have remembered to set the supervisor option to TRUE.

omichel, 2020-06-16:
The problem is probably because your Robot node doesn't have a DEF name equal to "MY_ROBOT". Please check the scene tree and ensure your Robot has a correct DEF name.

closed 2020-06-16 by MichaelFrancis123