cyberbotics/webots #1789: Crash When Inserting PROTO on develop

DavidMansolino, 2020-06-16:
**Describe the Bug**
When inserting the following PROTO node in the simulation with the add node dialog the simulation crashes with this assertion:
```
webots-bin: nodes/utils/WbTriangleMesh.cpp:418: void WbTriangleMesh::indicesPass(const WbMFVector3*, const WbMFInt*, const WbMFInt*, const WbMFInt*): Assertion `mCoordIndices.size() == mTmpNormalIndices.size() || mTmpNormalIndices.size() == 0' failed.
./webots: line 88: 126293 Aborted                 (core dumped) "$webots_home/bin/webots-bin" "$@"
```

[SpacePirateV5_2.zip](https://github.com/cyberbotics/webots/files/4786813/SpacePirateV5_2.zip)

On master it is working fine.

DavidMansolino, 2020-06-18:
The problem comes from the normals, it seems there are more normals than coordinates.

omichel, 2020-06-25:
Can we close this issue? It seems to be fixed in #1813.

DavidMansolino, 2020-06-25:
Yes sure, it seems it was not closed automatically.

PR #1813 (2020-06-22): Fix Crash with Normals
**Description**
Fix #1789
closed 2020-06-25 by DavidMansolino