cyberbotics/webots #1794: Broken IndexedLineSet Appearance

DavidMansolino, 2020-06-17:
**Describe the Bug**
The appearance of the indexedLineSet node is not working, it is always black:
[image]
Here is how it should looks like (image of the doc):
[image]

omichel, 2020-06-18:
I suppose this regression was introduced when we generalized the PBRAppearance.
This example uses PBRApperance for the IndexedLineSet which doesn't really makes sense.
If we replace the PBRAppearance with an Appearance node with the ambiantIntensity set to 1, the color of the IndexedLineSet appears to be correct.

DavidMansolino, 2020-06-18:
It seems you are right, I checked and it seems it never worked with PBRAppearance.
But we should in any case either fix this either deny the use of PBRAppearance in the appearance field of Shape node using an indexedLineSet.

PR #2142 (2020-08-26): Fix unsupported PbrAppearance in IndexedLineSet and PointSet
Fixes #1794.
closed 2020-08-26 by omichel