cyberbotics/webots #1798: "white noise" is not equivalent to "uniform noise"

goretkin, 2020-06-17:
**Documentation page(s) concerned**
https://github.com/cyberbotics/webots/blob/9681a5707563a947a837edfa9997393049be7444/docs/reference/differentialwheels.md#L60

**What is the problem with the current documentation?**
"White noise" has a uniform _power spectrum_, not a uniform distribution per sample. Uniform density noise is white noise, but so are many other distributions.  See https://en.wikipedia.org/wiki/White_noise

**Describe how you would fix it**
Remove reference to white noise, since white noise is not equivalent to uniform-density noise.

closed 2020-06-18 by omichel

DavidMansolino, 2020-06-18:
From the page you are citing:
> the pixels of a white noise image are typically arranged in a rectangular grid, and are assumed to be independent random variables with uniform probability distribution over some interval

This seems to be exactly our case no?

goretkin, 2020-06-18:
> From the page you are citing:
> 
> > the pixels of a white noise image are typically arranged in a rectangular grid, and are assumed to be independent random variables with uniform probability distribution over some interval
> 
> This seems to be exactly our case no?

`The noise has a uniform distribution, also known as as "white noise."`

Looks like it was fixed. Thanks!

PR #1799 (2020-06-18): Fixed noise description
Fixes #1798.