cyberbotics/webots #1804: Random physics issue in a zero gravity environment

Acwok, 2020-06-18:
**Describe the Bug**
I'm moving a robotic arm in zero gravity.  
I noticed that sometimes the arm drifts while it should stop moving as soon as it finishes executing its trajectory. I could reproduce this situation on the UR10 robot provided in Webots.  
Here is a screencast showing the issue: https://youtu.be/S7VO7TUH4pk?t=475  
Here is another with my custom arm: https://youtu.be/goqD1SOuOnk?t=560
It seems to happen randomly. I've also encountered this on the very first executed trajectory but I wasn't recording.

**Steps to Reproduce**
- Follow the https://cyberbotics.com/doc/guide/ure tutorial to setup MoveIt for the Universal Robot UR10.
- create a scene with no gravity and basictimestep set to 8ms
- add an UR10 robot with static base to false and its controller set to `universal_robots_ros`
- plan and execute trajectories using moveit

**Expected behavior**
The arm executes the trajectory and stops moving.

**System**
- ROS Melodic
- Ubuntu 18.04 in a docker with GPU acceleration
- Webots build from sources commit `fe5893b968e160447a52cc4619ad68a8f41f8022`

DavidMansolino, 2020-06-19:
Thank you for this precise bug report.
To help us better identify the problem, would it be possible for you to create a simple controller (pure Webots API) that reproduce the issue?
I tried and I am not able to reproduce any issue with 0 gravity with the sample simulation provided in webots for the universal_robot robots.

Acwok, 2020-06-19:
Hi @DavidMansolino,
I've made a test using pure webots API (here are the sources: [ur10_test.zip](https://github.com/cyberbotics/webots/files/4805365/ur10_test.zip))
The bug is still there, see https://youtu.be/y37loILy_1I    

I used the ure sample world, set gravity to 0, robot static base to FALSE, enabled self collision.  
The controller applies random positions between [-1, 1] on motor 0, 1, 2 (not 3) to avoid collisions.  
You will also notice in the video that  some bounding boxes turn to red while there is not collision.

DavidMansolino, 2020-06-22:
Why did you set the static base to FALSE? 
About the fact that is doesn't stop drifting, this seems to make sense to me, there is nothing that is slowing down the robot (no friction), if you want it to stop, you should probably add some `Damping` (https://cyberbotics.com/doc/reference/damping) to the worldInfo (https://www.cyberbotics.com/doc/reference/worldinfo) node.

Acwok, 2020-06-22:
> Why did you set the static base to FALSE?

The goal is to use a robotic arm fixed on a spacecraft to assemble large structures in space.  
Here is a short video presenting the project H2020 PULSAR: https://youtu.be/tMvbsFrNOpU.
The arm cannot be fixed to the environment because in reality any action from the arm should induce a counter reaction on the spacecraft.  
One part of the project is to create a controller to stabilize the spacecraft so that it always points in the same direction.

> About the fact that is doesn't stop drifting

Yes with no friction, it will continue drifting, the problem is that it shouldn't start drifting.  
In the second video (https://www.youtube.com/watch?v=goqD1SOuOnk&feature=youtu.be&t=560), the arm moves then stops during 10 minutes (which is good I think), and suddenly it starts drifting a lot.

Maybe it is due to:
- Internal collisions. Indeed, some bounding box turn red in the UR10 video while there is no apparent collision. However, I haven't noticed this on our custom arm.
- Something with the ERP or CFM parameters ?
- An issue with the motor PID or `wb_motor_set_position()`

DavidMansolino, 2020-06-22:
You right, it should indeed not drift.
I am suspecting internal rounding issues that are causing this problem.
Self-collision might indeed be one of the causes of the drift, have you tried disabling it? Have you tried also to reduce (drastically to test) the timestep of the simulation?

Acwok, 2020-06-22:
Disabling self-collision doesn't solve the issue, however removing the bounding objects does.
Even with a 32ms timestep.

Reducing the timestep to <=8ms also seems to work (without removing the bounding objects nor the self-collision parameter).  

So it would be a collision issue ?

DavidMansolino, 2020-06-23:
In the non-bounding object case, are you defining the physics mass, center of mass and inertia matrix?

Acwok, 2020-06-23:
I've just converted the UR10 proto to base node and removed the bounding objects.  
The physics fields stay the same, that is to say:
```
density -1
mass 13.051 (depends on the joint)
centerOfMass [0 0.3065] (depends on the joint)
``` 
When openning the world a message says all inertia matrices are set to identity:
```
WARNING: Robot > Robot: Undefined inertia matrix: using the identity matrix. Please specify 'boundingObject' or 'inertiaMatrix' values.
WARNING: Robot > Robot > HingeJoint > Solid: Undefined inertia matrix: using the identity matrix. Please specify 'boundingObject' or 'inertiaMatrix' values.
WARNING: Robot > Robot > HingeJoint > Solid > HingeJoint > Solid: Undefined inertia matrix: using the identity matrix. Please specify 'boundingObject' or 'inertiaMatrix' values.
WARNING: Robot > Robot > HingeJoint > Solid > HingeJoint > Solid > HingeJoint > Solid: Undefined inertia matrix: using the identity matrix. Please specify 'boundingObject' or 'inertiaMatrix' values.
WARNING: Robot > Robot > HingeJoint > Solid > HingeJoint > Solid > HingeJoint > Solid > HingeJoint > Solid: Undefined inertia matrix: using the identity matrix. Please specify 'boundingObject' or 'inertiaMatrix' values.
WARNING: Robot > Robot > HingeJoint > Solid > HingeJoint > Solid > HingeJoint > Solid > HingeJoint > Solid > HingeJoint > Solid: Undefined inertia matrix: using the identity matrix. Please specify 'boundingObject' or 'inertiaMatrix' values.
WARNING: Robot > Robot > HingeJoint > Solid > HingeJoint > Solid > HingeJoint > Solid > HingeJoint > Solid > HingeJoint > Solid > HingeJoint > Solid: Undefined inertia matrix: using the identity matrix. Please specify 'boundingObject' or 'inertiaMatrix' values.
```
Here is the world and controller: [ur10_no_bounding_box.zip](https://github.com/cyberbotics/webots/files/4818227/ur10_no_bounding_box.zip)

DavidMansolino, 2020-06-23:
OK, in that case probably the point is that it use the identity inertia matrix, and therefore simplify a lot the dynamic of the arm (which probably removes these small instabilities causing the drift).

In any case, a time step of 8ms is not so bad so I would simply go for the use of a 8ms time step.

Acwok, 2020-06-23:
In the second video: https://youtu.be/goqD1SOuOnk?t=560, with our arm, the basic time step was already 8ms and the problem occured. I hope that using a smaller timestep will do the trick :crossed_fingers: .   

Also, the inertia matrices were computed by Webots using the bounding box, maybe I should set them manually and use more accurate values before tweaking the timestep.

DavidMansolino, 2020-06-23:
About the inertia matrices, I would also try to make them diagonal (keep only the principal components).

omichel, 2020-11-30:
I am closing this stalled issue. Feel free to re-open if needed.

closed 2020-11-30 by omichel